armel/uv-k5-firmware-custom-feat-F4HWN #1: Ultra low power setting

ON7MHZ, 2024-01-15:
Hi , 
not sure if its an issue or wrong manipulation on my side ...
Is it possible that the ultra low power (like 125 mW) is only available in VFO mode? 
If I adapt the power in the menu 61/61 on a memory channel, and reboot the device it seems to be back at the original power setting.

And suggestion ... in the IJV firmware you can see the output power on the display, maybe something that van be integrated in an next update?

Keep up the good work,really love playing with this firmware ! 

73 de ON7MHZ

armel, 2024-01-16:
Hello,

Whether in VFO or on a memory channel, if you are in Low power mode, the power sent will be that selected from Set Low menu option. Be aware, after changing Low level with Set Low menu option, you need to restart your pocket !
About display the output power directly on the screen, it's an idea. But I'm not sure to find a free space somewhere. I'll have a look.

73 de F4HWN.

joc2, 2024-01-18:
suggestion, add a option in SetLow, OFF,,, so if SetLow is off, then the power will be the one set in egzummer eeprom, else, take the one select in the SetLow...

maxibor, 2024-02-16:
Hey @armel ,
Thanks for cool work on the UV-K5 firmware :)
Did you measure the power outputs, or are they simply estimated ?
Also, what is the reason behing the `Txp[p] = (Txp[p] * 4) / X` instead of just dividing by `Y` ? 

Thanks again for the firmware mods !

armel, 2024-02-16:
Hi,

Yes, I've made several videos on this subject, on my Youtube channel. 
And yes I took measurements with a simple Diamond SWRmeter and a TinySA.
About your question concerning the code, this allowed me to be more precise to tune it. 

Regards,

Armel.