armel/uv-k5-firmware-custom-feat-F4HWN #2: nice option low power and the / instead of the .

joc2, 2024-01-18:
it's not a issue... but a thumbs up 

i push a request to egzumer and put you as the idea comming from....   
nice thing.... 👍 

i am working on the chirp driver for egzumer firmware  in egzumer repo

do you have the code, i want to know where do you store the info in memory, eeprom, flash.... 
at what address....

armel, 2024-01-18:
Hello,

The code of my modifications is available here https://github.com/armel/uv-k5-firmware-custom/tree/feature_f4hwn

Armel F4HWN.

joc2, 2024-01-20:
I made a version driver for chirp for your modif for the setlow... i test it and it work... :) 
if your interest to it. let me know,  send me a address where i can send it to you.

closed 2024-02-08 by joc2