armel/uv-k5-firmware-custom-feat-F4HWN #3: Sensibilité Squelch

F5GEB, 2024-01-28:
Bonjour, c'est une issue générale à tous les firmwares : Le seuil de déclenchement du squelch est réglé trop haut .
Est-il possible de changer  la valeur correspondante à squelch = 1 afin de permettre de déclencher sur des signaux plus faible ? En effet   des signaux  faibles sont parfaitement audible quand le squelch est ouvert manuellement (ou squelch = 0 sur des signaux aviation ou satcom), mais pas avec squelch = 1 merci et 73 de Eric F5GEB

armel, 2024-01-29:
Bonsoir,

Oui, le réglage du Squelch semble poser des problèmes chez quelques OM. Je regarderais à l'occasion, mais c'est une partie du code que je n'ai que survolé. Je dois m'y plonger plus en détail.

73'

Armel F4HWN.

F5GEB, 2024-01-29:
Merci, c'e n'est pas la sensibilité (donc les valeurs de gain) qui pose problème, mais bien la valeur de seuil de déclenchement. 

Sur le firmware IJV, il y a des valeurs en dB affichées pour chaque niveau de squelch mais le problème reste le même, le seuil pour la valeur "Sql=1" est trop haute, ci-dessous les valeurs pour info :

Sql=0 : -143dB (Squelch ouvert)
Sql=1 : -125dB 
Sql=2 : -124dB 
Sql=3 : -123dB 
Sql=4 : -123dB 
Sql=5 : -122dB 
Sql=6 : -121dB 
Sql=7 : -120dB 
Sql=8 : -119dB 
Sql=9 : -119dB

Je ne sais pas quelles sont les valeurs sur Egzumer, donc je me réfère à ce que je vois sur IJV : l'écart entre 0 et 1 est de 18 dB puis seulement de 1dB pour les autres pas.. descendre le seuil de Sql=1 au alentour de -133dB (plus ou moins en fonction du seuil de bruit) puis incrémenter de 1 dB pour les pas suivant me semblerait meilleur.

Un autre possibilité serait de faire (comme pour la puissance LOW) plusieurs valeurs sélectionnable pour la valeur de Sql=1 et ne pas toucher aux autres valeurs de Sql=2 à Sql=9.

J'éspère que mon analyse est claire et encore merci d'y jeter un œil ;)

73, Eric F5GEB

LHunti, 2024-01-30:
The values ​​in the squelch table write might solve the problem: https://github.com/rebezhir/openquack/tree/main/squelch-tuner

armel, 2024-02-16:
Bonsoir Eric,

Je vous encourage à utiliser le driver Chirp spécifique à ma version de firmware et de modifier les valeurs de calibration dédiées au Squelch. J'ai publié, au cas ou, une petite vidéo rapide sur ce driver Chirp dédié et j'y aborde le sujet du Squelch également. C'est dispo sur ma chaine Youtube (https://www.youtube.com/watch?v=02T2ODufZOA). 

Au préalable, je vous recommande de faire un backup de votre Eeprom.

73'

Armel F4HWN.

closed 2024-02-20 by armel