armel/uv-k5-firmware-custom-feat-F4HWN #4: LED Flicker on Rx

KDMcMullan, 2024-02-09:
Really like this fork.
When I have BLmax set to anything other than 10, the LED flickers to full brightness at the start of Rx. Is this normal?
Thanks.
NB: Personally, I think it would make a great optional feature: flash BL on Rx!

armel, 2024-02-09:
Sorry, I can't see the link between the BLmax (max screen brightness setting) and the LED.
By the way, which led are you talking about? The big white led on top of the transceiver or the little led at the base of the antenna ?

KDMcMullan, 2024-02-09:
Apologies: I meant the LED backlight. If I dim the backlight (BL to any setting below 10) the backlight flickers when an incoming signal begins.

However, I've just spotted _"improve UI (menu, s-meter, MAIN ONLY screen, **RX blink,** RX LED blink, Squelch level, etc.),"_ so I assume the backlight has been specifically programmed to do this.

armel, 2024-02-09:
Oh, very strange. So no, this is not normal behavior.

closed 2024-02-20 by armel

KDMcMullan, 2024-03-11:
For info, I think this has been fixed. I'd completely forgotten about it, but the latest build doesn't exhibit the behaviour.
I actually sort of miss it and wonder if it would make a good feature!!