armel/uv-k5-firmware-custom-feat-F4HWN #6: Scrambler activated after full reset (v1.6 and v1.7)

ON7MHZ, 2024-02-09:
Hi , 

While using my quansheng today (on v1.6) I suddenly lost RX audio on the internal speaker (while listening to VHF airband).
No more sound, no more beep , ... completely silent.
When connecting to an external mic/speaker everything was fine.
I tried a bit of everything, not being sure is this was a hardware or software issue. 
I performed a full reset via (secret) menu 67/67 -> ALL -> confirm.
Handheld rebooted with still no audio... but the (removed) scrambler function was activated (on RX and TX).
Updated my device to v1.7
Now I have audio again (woohoo!), but SCRAMBLER is still active (and no way to de-activate via the menu (anymore)) 
Also some strange indication below the upper VFO freq (see pic)
Tried an other reset on v1.7 ( both VFO only and FULL ) but no change and SCR still active.
So I will downgrade (temporarily) to an older version to deactivate the scrambler) 
[image]

PS : since I initially thought it was a hardware issue I ordered a new UV-5R plus , so now you own me 18 euro ;)  HI

armel, 2024-02-09:
SCRAMBLER functionality has been removed from the standard menu in V1.6. I also removed it from the hidden menu in V1.7. Performing a full reset under V1.6 (and V1.7) probably reactived it. I will take a look...

ON7MHZ, 2024-02-09:
That was a quick reply! 

After some more playing...

Downgraded my UV-5Rplus to v1.5 
Scrambler function was automatically disabled again, so there was no extra action needed beside the downgrade and switching on the device. Checked the menu and could enable/disable SCR without any issue ( only played on upper VFO)
Haven't done or tried anything else beside SCR function.

Updated back to v1.7 

Scrambler active again automatically actice on lower VFO, upper VFO working normally this time.
Again the 'strange' indicator when selecting lower VFO, it looks like a 'mix' of "SQL 2" and "SCR" on top of each other ( see pics)

And keep up the good work ! 73
[image]
[image]

armel, 2024-02-16:
Hi,

Could you try v1.8 ? I clean the code more deeply about SCRAMBLER.

Armel.

closed 2024-02-20 by armel