armel/uv-k5-firmware-custom-feat-F4HWN #7: Led S.O.S and Blink

tacva, 2024-02-14:
Il ne s'agit pas d'un dysfonctionnement, mais d'une proposition de modification. Si cette demande est mal placée, je m'en excuse. Je voudrais proposer de supprimer le clignotement de la LED lors de l'utilisation de la lampe de poche, en laissant uniquement la fonction marche/arrêt. Serait-il possible de le faire ? Le jugez-vous utile ?

tacva, 2024-02-16:
I think the code part is this one, in flashlight.h

#ifndef APP_FLASHLIGHT_H
#define APP_FLASHLIGHT_H

#ifdef ENABLE_FLASHLIGHT

#include <stdint.h>

enum FlashlightMode_t {
    FLASHLIGHT_OFF = 0,
    FLASHLIGHT_ON,
    FLASHLIGHT_BLINK,
    FLASHLIGHT_SOS
};

extern enum FlashlightMode_t gFlashLightState;
extern volatile uint16_t     gFlashLightBlinkCounter;

void FlashlightTimeSlice(void);
void ACTION_FlashLight(void);

#endif

#endif

armel, 2024-02-16:
Hello,

Of course, it's possible. But the question is why ?
The FlashLight action lets you choose between several states: on (fixed), flashing, SOS or off. 
Each of these states has its own purpose. 

Armel.

tacva, 2024-02-17:
First of all, thank you for your reply. The reason for my suggestion is that I find it tedious, once the LED is activated, to have to press and hold the button multiple times just to turn it off. First press: blinking, second: SOS, then finally it turns off. How often do I use the flashlight on the radio: a few times. How many times in my entire life have I used the blinking and even worse the SOS? Never :-) However, I admit that if I am the only one who feels this need, it is likely that the issue is of little importance ;-) May I ask if it is possible to download an uncompiled version of your firmware?

armel, 2024-02-17:
Okay, thank you. I understand. 
Well, if I have several similar requests about the LED, I'll do it. No problem ;)

closed 2024-02-20 by armel