armel/uv-k5-firmware-custom-feat-F4HWN #8: Question: differences between v1.9 and v1.9b

KerseyPaul, 2024-02-19:
Hi there Armel!

As per the descriptions provided in the releases section, it wasn't clear what exactly was correted/implemented in version 1.9b when compared to the previous 1.9. Could you please clarify?

Thanks in advance!

armel, 2024-02-20:
Hello,

Yes, of course.
On version 1.9, there was a bug with LOW powers. 

LOW1 = 125mW
LOW2 = 250mW
LOW3 = 500mW
LOW4 = 1W
**LOW5 = <20mW**

On version 1.9b, 

**LOW1 = <20mW**
LOW2 = 125mW
LOW3 = 250mW
LOW4 = 500mW
LOW5 = 1W

In addition, version 1.9b also displays the CTCSS or DCS code, below the frequency, in Dual VFO mode (and if SetGui is set to Tiny).

Regards,

Armel F4HWN.

spamsool, 2024-02-20:
I tried to display CTCSS code in PMR band and I could not. No code shows below frequency. Am I missing anything?

armel, 2024-02-20:
Ok, so yes, you certainly miss something.
First of all, there's CTCSS (or DCS) for reception (Rx) and CTCSS (or DCS) for transmission (Tx).

Try this on your transceiver.

1 - move to a PMR memory channel
2 - enter menu 4, **RxCTCS**, choose (for example) 67.0 Hz and confirm
3 - now take a look, CT 67.0 is visible on the VFO screen !

Now, if you select a **TxCTCS** with menu 6, choose (for example) 71.9 Hz and confirm

You always see CT 67.0 on the VFO screen, because your transceiver is on Rx. But if you push PTT, tour transceiver in on Tx and you will see CT 71.9 Hz...

spamsool, 2024-02-20:
Thank you very much for the extensive explanation.

closed 2024-02-29 by armel