armel/uv-k5-firmware-custom-feat-F4HWN #10: Very low screen contrast after v0.19b update

Daffouille, 2024-02-21:
Hello

The screen is barely visible after this update from Original Egzumer 0.22 => EGZUMER feat F4HWN v1.9b

(I also speak French ... just in case ...)

[image]
[image]

armel, 2024-02-21:
Hello,

Yes, it's normal. Use Menu SetCtr to set contrast.

Armel.

Daffouille, 2024-02-21:
Hello,

Oui, mais c'est perturbant

Il faudrait peut être penser à le spécifier dans la doc utilisateur avec le nom du menu ou aller chercher ce réglage de contraste (ou dans la release note), tout comme rappeler la procédure de reset VFO après le chargement du firmware (même si c'est évident pour vous).

On ne prend jamais assez soin des utilisateurs ;)

Cdlt

armel, 2024-02-21:
Difficile de tout maitriser, de savoir quel firmware était installé avant le mien (sans compter qu'il y en a tellement), de connaitre l'état de l'eeprom, etc. Du coup, chez certains, au premier démarrage, le contraste est effectivement trop bas, j'en suis conscient. 

Maintenant, concernant les utilisateurs. C'est une façon de voir les choses. Ceci dit, je ne suis pas du tout dans une démarche _produit_ et je ne vise aucun profit. C'est toujours bon de le garder en tête ;) 

Je développe ce firmware avant tout pour moi et mes usages. Et je le mets, au cas ou, à disposition de la communauté sans rien attendre en retour. En complément, j'annonce la couleur dans introduction du README avec la mention  "About this **experimental** version feat by F4HWN" :)

Le mot **experimental** a tout son sens ici ;) 

Armel.

Daffouille, 2024-02-21:
Oui, c'est clair, et c'est déjà super comme cela, j'imagine le boulot, mais aussi le plaisir à creuser un tel sujet.

Stéphane

closed 2024-02-21 by Daffouille