armel/uv-k5-firmware-custom-feat-F4HWN #11: v1.9b : Plage du "Scan en fréquence" incomplète

Daffouille, 2024-02-21:
Hello

Je pense que ceci est un réel bug par contre (aussi présent dans la v0.22 EGZUMER).

VFO A : 446.00625 (PMR Canal 1)
VFO B : 446.09375 (PMR Canal 8)

[image]

Si je passe en mode Scan (Appui long sur 5) ...

[image]

... et que je lance le Scan avec un appui long sur *, tout semble bien fonctionner.

Sauf que le Scan s'arrête bien si je pompe avec un autre portable sur les canaux PMR 1, 2, 3, 4, 5, 6, 7 ... mais le Scan ne s'arrête pas si je pompe sur le Canal 8 ...

Cdlt

armel, 2024-02-21:
Ah ok, autant pour moi, le canal 8 est bien sur 446.09375. Il faut dire que je n'y vais pour ainsi si dire jamais, sauf pour 2 ou 3 tests.

Peut-être qu'étant en limite de plage, le ScanRange ne le prend pas en compte ?
En terme mathématique, le ScanRange couvre peut-être [446.00625, 446.09375[ (intervalle ouvert) et non [446.00625, 446.09375] (intervalle fermé).

L'utilisation des listes reste tout de même l'idéal, de mon point de vue.

Armel.

Daffouille, 2024-02-22:
Que ce test soit fait la bande PMR ou bien sur une autre, le comportement restera le même.

D'un point de vue utilisateur, un tel comportement lui semblera illogique car il s'attendra à un même fonctionnel aux 2 bornes de son ScanRange (ce qui n'est actuellement pas le cas, car seule la borne inférieure est prise en compte).

Comme vous le dites, nous avons surement à faire à un "intervalle ouvert" et non fermé (surement une sombre histoire d'< au lieu d'<= dans le code, ou quelque chose de similaire). 

Cela dit, je remonte juste l'info, c'est tout.

Il y a plusieurs façons de contourner le phénomène, l'utilisation des listes en est une en effet.

Stéphane

armel, 2024-02-22:
Bon, j'ai jeté un oeil et c'est bien ce que je pensais :

```
	if (Frequency >= upper)
		Frequency =  lower;
```

Donc, quand le scanRange atteint votre limite haute de 446.09375, il retourne sur la limite basse 446.00625.
Je peux prévoir de le fixer dans la prochaine release qui sortira... je ne sais pas quand.
Il suffira de faire ceci,

```
#ifdef ENABLE_FEAT_F4HWN
	if (Frequency > upper)
#else
	if (Frequency >= upper)
#endif
		Frequency =  lower;
```

Et c'est plié.

Daffouille, 2024-02-22:
Si ca coute trop en taille de FW, il faut clairement oublier ...

armel, 2024-02-22:
Non non, rassurez-vous, ce sera totalement sans conséquence sur la taille du firmware.

Daffouille, 2024-02-22:
Alors let's go !!!!!!!!!!!!!!!!!!!!!

Daffouille, 2024-02-22:
Tout bug doit être fixé, si cela ne coute pas trop cher

Daffouille, 2024-02-22:
L’expérience/confort utilisateur est votre (et mon, dans mon domaine) cœur de métier ;)

armel, 2024-02-29:
Correction effectuée dans le version v2.0

closed 2024-02-29 by armel