armel/uv-k5-firmware-custom-feat-F4HWN #12: Improvement, additionnal functionnality for SCAN mode

F6EGK, 2024-02-21:
Bonsoir Armel,

Et merci pour vos efforts dans ces compléments apportés à la version développée par Egzumer. 

Dans le contexte du projet OpenGD77, j'utilise une fonctionnalité que je trouve très pratique. Il s'agit lorsque le récepteur est en recherche (SCAN) de pouvoir temporairement exclure du balayage une ou des mémoires issues de la liste courante. Concrètement si le balayage s'arrête sur une mémoire où un qso est en cours, on peut exclure cette mémoire en appuyant sur une touche. Le balayage repart aussitôt, mais cette mémoire est ensuite ignorée dans les cycles suivants de balayage (SKIP). Cette procédure peut être répétée à l'identique pour exclure temporairement des mémoires supplémentaires. La sortie volontaire du mode SCAN, ou l'arrêt du TRx (OFF) annule le ou les choix précédents. Si le mode SCAN est relancé, l'intégralité des mémoires de la liste est à nouveau balayée.

Je ne sais pas si l'espace mémoire encore disponible permettrait d'implémenter cette fonctionnalité, c'est juste une suggestion !

For english readers 

Thank you for your efforts in pushing new features to the fimware version developed by Egzumer.

Coming from the OpenGD77 project, I use a feature that I find very practical. When the receiver is in searching mode (SCAN), it is possible to temporarily skip one or more memories from the current list. It works like this : if the scanning stops on a memory where a qso is in progress, this memory can be excluded by pressing a key. The scan restarts immediately, but this memory is then ignored in next scan cycles (skipped). This procedure can be repeated in a same way in order to exclude additional memories. Exiting from SCAN mode or turning off the TRx, cancels the previous choice(s). If a new SCAN is started, all memories in the list are normally scanned.

I don't know if the available memory space would allow a such functionality to be implemented, it's just a suggestion!

73 - Jean-Roger / F6EGK

reminoo, 2024-03-23:
Bonjour Armel, Bonjour Jean Roger,

Se serais une fonction très intéressante, 
De plus, pouvoir l'actionner via un raccourcis tel que "press short F1" se serais vraiment pratique. 

Lorsque j'utilise CHIRP, dans l'onglet Mémoire, colonne: Sauter. 
Je sélectionne la liste déroulante, il n'y a pas de proposition.
Tel que le manuel de CHIRP le décrit.
Le mode de saut de balayage pour la chaîne. Les valeurs et leurs significations sont :
S : ignorer cette chaîne pendant le balayage
P : Cette chaîne est prioritaire

Je vous remercie pour le partage des efforts afin d'améliorer le firmware et le driver CHIRP. 

Je vous salut bien cordialement.

armel, 2024-03-28:
Bonsoir,

Je viens de publier la version v2.4.
Il est désormais possible d'exclure des canaux mémoire, via un appui LONG sur la touche M (Menu),  pendant un scan sur la liste I ou liste II (ne fonctionne pas si scan ALL sur l'ensemble des canaux mémoire). 

Je cloture cette issue.

73'

Armel F4HWN.

closed 2024-03-28 by armel

F6EGK, 2024-03-29:
Bonjour Armel et félicitations pour l'intégration de cette fonctionnalité, c'est tout simplement génial ! C'est d'autant plus fort que vous êtes aux limites extrêmes de l'espace mémoire. Dans la dernière vidéo vers 13'20", vous mentionnez que l'action M long est maintenant réservée à cette fonctionnalité, et qu'une affectation antérieure est caduque. C'est peut-être uniquement vrai pour une fonctionnalité qui aurait été en lien avec les listes 1 ou 2 (ce point est peut-être ambiguë), car de mon côté l'affectation antérieure M long au menu 53 (Rx Mode) continue à être opérationnelle (hors scan liste 1 ou 2 bien sûr). Je chipote, c'est un pur détai !

Encore une fois, avec tous mes remerciements pour vos développements.

73 - Jean-Roger

armel, 2024-03-29:
Bonjour Jean-Roger,

Merci pour votre message. Oui, je l'ai précisé dans un commentaire et effectivement, je n'ai pas été assez explicite. Hors scan LIST 1 ou LIST 2, l'appui M LONG reste fonctionnel, comme auparavant, pour exécuter une action. Il l'est également dans le cas d'un scan ALL (puisque l'exclusion ne fonctionne pas sur ce type de scan).

73'

Armel F4HWN.