armel/uv-k5-firmware-custom-feat-F4HWN #14: SetLck configuration and down side key behavior

F6EGK, 2024-02-26:
Bonjour Armel,

C'est encore moi ...

La configuration par défaut de la touche latérale basse permet le passage en émission avec l'envoi de la tonalité 1750 Hz. Si dans le menu SetLck on active la configuration "Keys + PTT", cela ne modifie pas le comportement de cette touche latérale. Je pense qu'il ne devrait pas être possible d'activer le mode Tx lorque la configuration "Keys + PTT" du menu SetLck est choisie. Je ne sais pas si c'est assimilable à une anomalie, ou peut-être à une limite d'écriture de programme, matérielle, ou autre chose ?

73 - Jean-Roger

For english readers

Default configuration for down side key enables Tx + 1750 Hz tone sending, thanks to a short press. When SetLck is set as "Keys + PTT", this side key function remains available. My feeling is that it should not be possible to activate Tx mode in this situation. Do we consider this behavior as a bug, hardware issue, or something else ?

armel, 2024-02-26:
Bonjour Jean-Roger,

Historiquement, le verrouillage du K5 ne concernait que les 12 touches du clavier (touche de 0 à 9 + * Scan + F). J'y ai donc ajouté le fait de pouvoir également verrouiller le PTT, afin d'éviter des passages en émission intempestifs. 

Néanmoins, les touches M, F1, F2, Exit etc. restent toujours actives. Du coup, si vous avez paramétré un raccourci pour envoyer une tonalité 1750 Hz (par exemple sur F1), en l'état, il est normal qu'elle soit émise (même en étant en "Keys + PTT"), le vérouillage ne concernant pas ces touches. 

Ceci étant, votre demande me semble tout à fait légitime (et logique).

La modification sera donc effectuée et disponible dans la version 2.0. Il sera impossible d'envoyer la tonalité 1750 Hz, via un raccourci, si nous sommes en mode "Keys + PTT".

73' de F4HWN.

F6EGK, 2024-02-26:
Merci Armel pour vos informations, c'est très clair. Bons développements pour la communauté, d'autant que j'avais relevé la polémique stérile au sujet des différents firmwares disponibles, et leurs "paternités" supposées. Les principes du "fork"  n'ont visiblement pas été comprises (si tant est qu'ils aient cherché à comprendre !) ...

Bonne soirée et 73.

armel, 2024-02-29:
Correction effectuée dans la version 2.0

closed 2024-02-29 by armel