armel/uv-k5-firmware-custom-feat-F4HWN #19: S meter

Linkos0, 2024-03-01:
Hello, 
I installed your software and I have a question: are the S meter parameters correct?
Signal level indication without antenna seems incorrect.
I am attaching a photo

[image]
[image]

armel, 2024-03-01:
Hi,

Ok. you're confusing HF and VHF / UHF bands.
Your table are referring to HF bands.

HF S-meter

```
S9 + 40 dB	-33 dBm
S9 + 30 dB	-43 dBm
S9 + 20 dB	-53 dBm
S9 + 10 dB	-63 dBm

S9  -73 dBm
S8  -79 dBm
S7  -85 dBm
S6  -91 dBm
S5  -97 dBm
S4 -103 dBm
S3 -109 dBm
S2 -115 dBm
S1 -121 dBm
```

VHF/UHF S-meter is different...

```
S9 + 40 dB	-53 dBm
S9 + 30 dB	-63 dBm
S9 + 20 dB	-73 dBm	
S9 + 10 dB	-83 dBm

S9  -93 dBm	
S8  -99 dBm
S7 -105 dBm
S6 -111 dBm
S5 -117 dBm
S4 -123 dBm
S3 -129 dBm
S2 -135 dBm
S1 -141 dBm
```

Take a look here for more information : https://hamwaves.com/decibel/en/

Regards,

Armel F4HWN

SQ9PXB, 2024-03-02:
Hello

Will you share the source code for your part of the modification? In my opinion, SMETR compliant with IARU guidelines for VHF/UHF is completely impractical. Personally, I prefer the original SMETR performance from exumer v0.22. Your other software is great, but I would like to change SMETR "to suit me".

Regards,
Mateusz SQ9PXB

SP9LEH, 2024-03-02:
Hi Armel, you are doing a great job with this firmware. However, I join your request to restore the s-meter to the parameters from exumer v0.22 The current version is good in lab conditions for very low noise receivers with ultra high sensitivities. For low-cost transceiver in my opinion it is not suitable.

Regards Leszek SP9LEH

armel, 2024-03-02:
Hi,

Yes of course, source code is full available on my [GitHub](https://github.com/armel/uv-k5-firmware-custom). Take a look at branch _feature_f4hwn_v0.22_ :)

In my opinion, calibration of the Egzumer SMeter is totally useless. That's why I've changed it. It's similar to what you'd see on a Yeasu FT-65, for example.

And it doesn't change the fact that if S9 in HF is -73dBm, it's -93dBm in VHF/UHF :)

Regards,

Armel F4HWN.

SQ9PXB, 2024-03-02:
Thank you for your answer. Unfortunately, I don't see the feature_f4hwn_v0.22 branch....

Regards,
Mateusz SQ9PXB

SQ9PXB, 2024-03-02:
It's OK now, I just can't find it, the GitHub repository was called "uv-k5-firmware-custom" and not "feature_f4hwn_v0.22". I've already downloaded the content, thank you :)

Mateusz SQ9PXB

armel, 2024-03-02:
As I said, my code is in a branch 😌

Armel F4HWN.

SQ9PXB, 2024-03-02:
Thank you again for your help and great software. For my convenience, I have already recompiled it to my settings. I see that this is already version 2.2 :) At the beginning I had a problem with stackoverflow of 76 bytes, but I turned off the unnecessary function and it compiled.

Regards,
Mateusz SQ9PXB

closed 2024-03-03 by armel