armel/uv-k5-firmware-custom-feat-F4HWN #20: v2.1 - Possible improvement for ONEPUSH configuration when reaching TOT limit

F6EGK, 2024-03-02:
Bonjour Armel,

Et merci infiniment pour cette nouvelle version 2.1. 

Ce que je rapporte ici n'est pas un bug, mais une possible amélioration lors de l'atteinte du TOT. Si SET PTT est configuré en mode PTT, l'atteinte du timeout entraine l'interruption de l'émission, et l'opérateur relâche naturellement le bouton PTT (passage en réception). Dans le cas où SET PTT est configuré en mode ONEPUSH, l'atteinte du timeout entraine l'interruption de l'émission, mais l'état apparent est toujours celui de l'émission (bargraphe Tx visible, mais pas de puissance RF bien sûr). Ne serait-il pas possible dans cette situation de forcer l'état ONEPUSH pour un retour automatique en réception, afin d'éviter un appui supplémentaire du bouton PTT ? 

73 - Jean-Roger

For english readers

When ONEPUSH mode is set thanks SET PTT menu, reaching TOT limit turn off transmission, but state remains as an active transmission (displayed Tx bargraph, but of course no RF power). For exiting this state, it is necessary for the operator to press once again the PTT button. For this situation, could it be possible to automatically reset ONEPUSH state, in order to retreiving receiving mode ?

armel, 2024-03-03:
Bonjour Jean-Roger,

Merci pour ton message.
Ok, ce sera corrigé dans la version 2.2.

73'

Armel F4HWN.

armel, 2024-03-04:
Implémenté dans la version v2.2

closed 2024-03-04 by armel