armel/uv-k5-firmware-custom-feat-F4HWN #21: Power Outpt  L / M / H    not  as reported

DL4SDR, 2024-03-04:
L = not measurable > i ThinK i must "see"   the Pout Needle  working ?
M =  VHF 2W  UHF  1.5W
H =  5 W with Battery  / 6.0W  with Battery eleminator
 Is there any possibility  programming  it in a subMenue  in the future ?

DL4SDR, 2024-03-04:
actual figured out  Menue 55  BUT  e.g. Set  to 1 W   = arround  250mW

armel, 2024-03-04:
First of all, hello... that's not necessarily an option. 

You've probably used other _exotic_ firmwares before, that has modified your calibration eeprom? My firmware, like others, makes a point of not altering theses datas. 

Here is my calibration data

```
// my eeprom calibration data 
//
// 1ED0 32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff  50 MHz
// 1EE0 32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff 108 MHz
// 1EF0 5f 5f 5f 69 69 69 87 87 87 ff ff ff ff ff ff ff 137 MHz
// 1F00 32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff 174 MHz
// 1F10 5f 5f 5f 69 69 69 87 87 87 ff ff ff ff ff ff ff 350 MHz
// 1F20 5f 5f 5f 69 69 69 87 87 87 ff ff ff ff ff ff ff 400 MHz
// 1F30 32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff 470 MHz
```

What is yours ?

If necessary, try to restore your calibration data as out of the box and try again. Trust me, my power calibration is correct. 

Armel F4HWN.