armel/uv-k5-firmware-custom-feat-F4HWN #23: scan partiel main only

f4axk, 2024-03-08:
slt Armel

Vue sur le groupe facebook UV-K5 FRANCE
Mode Main Only
Quand on utilise le scan partiel en VFO et que ensuite on veut passer en mode mémoires  ,
sans être sortie du mode scan partiel.
j'ai pu le constater

[image]
[image]

armel, 2024-03-08:
Bonjour Laurent,

Merci pour ce retour. Oui, j'arrive à reproduire ce bug d'affichage. 
Il m'avait été remonté par Pascal F4ICR, il y a quelques jours.
Je tenterais de le fixer mais... pas sûr que cela rentre.

73'

Armel F4HWN

f1ixx, 2024-03-10:
Bonjour

Comme deja dit, je ne suis tres tres loin d'etre un specialiste en progammation.
J'ai modifier dans le fichier ui/main.c - Ligne 646<br>
ancien code : <code>if (gScanRangeStart) { </code><br>
nouveau code : <code>if ( (gScanRangeStart) && (IS_MR_CHANNEL(gEeprom.ScreenChannel[vfo_num]) == false) ) { </code><br>
Il fort possible que j'ajoute plus de mal que de bien. D'apres les quelques essais, ca semble
fonctionner sur ma compilatio.

Merci

Boris

armel, 2024-03-11:
Le bug sera fixé dans la v2.3

Armel F4HWN.

closed 2024-03-11 by armel