armel/uv-k5-firmware-custom-feat-F4HWN #24: Which CHIRP version?

KDMcMullan, 2024-03-10:
Hi. I cannot get this to work with the new version of CHIRP. According to egzumer, the egzumer driver is now included in CHIRP but I'm getting "Firmware 'F4HWN v2.2' is not supported."
The F4HWN mod seems to work with egzumer's own CHIRP build.
Any thoughts what might be going on there?
Thanks.

f1ixx, 2024-03-10:
Hi

You can find a special nice version [here](https://github.com/armel/uv-k5-chirp-driver)
and have some help to use it.

Best regard
Boris

KDMcMullan, 2024-03-10:
Indeed, this is the one which works for me. Thank you.
HOWEVER, the question is: given that CHIRP and egzumer both say that the egzumer driver is included in CHIRP now, I wonder why CHIRP doesn't work with the F4HWN mod?

f1ixx, 2024-03-10:
Some function work with Egzumer, but haven't full function.
And Chirp use signature version to detect if driver is correct (eg 2.0 and 2.5).
F4HWN make many mods

KDMcMullan, 2024-03-10:
Je crois que CHIRP effectue des certaine validations et que le driver effectue également les autres.
J'ai remarqué que je ne peux pas non plus utiliser le nouveau CHIRP avec un egzumer "normale", puisque je ne suis pas autorisé (par CHIRP, j'imagine) a entrer des fréquences de bande de 11m.

armel, 2024-03-11:
Hi,

My firmware is experimental. As long as my firmware continues to evolve, you'll need to use the dedicated driver. If, and only if, my firmware stabilizes and becomes widely used (like Egzumer's), Chirp may choose to add it natively. But we're a long way from that. 

Now, note that using a dedicated driver is not complicated. It's just a little effort. 

73'

Armel F4HWN.

closed 2024-03-11 by armel

KDMcMullan, 2024-03-11:
I'm perfectly happy and comfortable to continue to use egzumer's driver. It's no chore at all. I'll also be continuing to use your firmware, as I really like the way it is shaping up.

I don't fully grasp the connections between the versions and their drivers. I present the following as facts:
- It's not clear that your users still need to use egzumer's driver.
- It's not clear why, even though egzumer's sriver IS now included in CHIRP, it still doesn't work.

I wonder if uptake of your build might be larger if folk weren't upset and put off my CHIRP telling them it doesn't work?

Hope this is helpful.