armel/uv-k5-firmware-custom-feat-F4HWN #25: Last signal received indicator

tacva, 2024-03-11:
Hi Armel,
congratulations again on your work.

I would like to propose an implementation; I'm not sure if others might find it useful.
It often happens to me to have the radio listening to two frequencies, and at some point the squelch opens on one of the two and I hear a transmission. After a moment, the transmission ends, I look at the display, but I have no way of knowing which of the two frequencies the reception occurred on. I keep the transmission frequency fixed on one of the two, so I have no indication because the radio does not automatically switch to the last received frequency. It would be useful to have an indication of the last received transmission on either the top or bottom frequency on the display, with any sign, character, or something similar.
Do you think it could be useful? (I hope I explained myself).

Thank you

armel, 2024-03-11:
Hi,

Ok. This depends on the memory available for development. At this point, it's too short. 

Armel F4HWN.

tacva, 2024-03-11:
I realize, I see that when you publish an update, the remaining memory gets closer and closer to zero. :-)
Thank you anyway. I find that your version, especially for the graphic layout of the display (but not only), is the best!

armel, 2024-03-14:
Add in v2.3

closed 2024-03-14 by armel