armel/uv-k5-firmware-custom-feat-F4HWN #28: Passage a main only en mode scan

fifisoft57, 2024-03-15:
bonjour

je viens de faire un test, j'ai lancer un scan sur la VFO B , j'ai programmer une touche F2 long pour changer de main only a normal

quand je changer et que le scan est en route , le scan continue en main only ( normal ) mais que je repasse sur l'autre affichage , le scan passe sur la VFO A et si je rappuies encore la ca passe sur la VFO B ( le scan continue )

a voir si  c'est normal ? 

merci pour se firmware qui est super et encore un super boulot

Philippe

armel, 2024-03-15:
Bonjour Philippe,

Laurent a parfaitement expliqué l'origine du phénomène. Je n'ai rien à ajouté :)

Armel.

fifisoft57, 2024-03-16:
merci bcp pour vos retour

closed 2024-03-16 by armel