armel/uv-k5-firmware-custom-feat-F4HWN #29: Minor Cosmetic issue, NOT important

F4VSJ, 2024-03-15:
I flashed to the 2.3 firmware
and I have settings like this:
POnMsg Sound
Rx Mode Main only
setptt onepush
Set Inv ON
Set Met Classic
when switching  the handheld ON there are some "white" pixels in the starting Screen
When I switch the PonMsg on All,  all is ok
See you Armel
73's de Hans F4VSJ

armel, 2024-03-15:
Hello Hans,

Yes, I know. I needed some free memory space to finalize this firmware. So I had to gain a few bytes here and there. This has no effect on the proper functioning of the pocket ;)

I will fix it if I remove FM Broadcast or something else :) 

Armel F4HWN

F4VSJ, 2024-03-16:
Armel, 
Oui oui le pocket marche bien avec to FW !
Un beau travail que tu as fait!!
Au pays bas , groupe pg540, on utilisr aussi ton FW , juste a cause du ptt onekey… et le bip bip en attendant sur les hotspots…

Et puis, je tombe juste sur les quelques bits “blancs” sur l’ecran, sans chercher…
Je trouves bien de voir la versipn de FW quand on allumes le pocket

Et OUI pour moi, d’enlever le radio fm, pour avoir de la memoire de libre.
Quand qqun veux ecouter le fm, bpn qu’il choisis in deuxieme quangcheng ou autre pocket, vus le prix du gadget.
J’en ai achete directement deux de 12 ou 13 Euros…
J’aimerais bien une fois jeter un coup d’oeil au dessus de tes epaules, quand tu changes dans le firmware…, comment faire…
Bonne programmation et a plus
73 de hans f4vsj

armel, 2024-03-16:
Bonsoir Hans,

Merci pour ton message.
Tu as bien raison. Ceux qui veulent écouter la bande FM n'auront qu'à acheter un second K5 :) Vu le QSJ, ca reste raisonnable. 
Concernant la programmation, il faudrait que je fasse une petite vidéo, en live sur Youtube, un jour. Ca pourrait être amusant :)

73' 

Armel F4HWN.

armel, 2024-03-28:
Hi Hans,

Fixed in the v2.4 firmware I just published :)

73'

Armel F4HWN.

closed 2024-03-28 by armel