armel/uv-k5-firmware-custom-feat-F4HWN #32: ZVEI selcall function.

baptistemichaud, 2024-03-17:
Good morning,
I would like to make selective ZVEI calls as is possible with the IJV firmware, to trigger mountain rescue (REGA, Codis 74). I looked for the functionality in your documentation, but I couldn't find one, is there one in this firmware?

I wish you very good luck, and good luck for the rest of this improvement.

armel, 2024-03-17:
Hello,

I've never used them. Are they used by radio amateurs? Use cases?

Armel F4HWN.

baptistemichaud, 2024-03-18:
Bonjour,  
Oui, elles sont utilisées par certains radioamateurs. Elles permettent de déclencher certains relais, ce qui permet de transmettre un identifiant. Le principe est plus ou moins le même que celui du DTMF, mais la différence est que ce ne sont pas les mêmes tonalités.

lien d'un site explicatif : http://www.ham-international.eu/pages/SC-100Z.html

enregistrement audio d'une tonalité ZVEI 1 ou 2 à 5 tonnes : https://www.swisstransfer.com/d/d7368576-3f25-481b-a0da-c34c875b5463

voici une réponse plus claire de chat gpt :

"Les tonalités ZVEI (Zentralverband Elektrotechnik und Elektroindustrie) sont un ensemble de tonalités utilisées dans les communications radio, notamment par les services de secours, et certains radioamateurs. Elles sont souvent utilisées pour la signalisation et le contrôle à distance. Les tonalités ZVEI sont caractérisées par leur simplicité et leur fiabilité dans la transmission des messages courts."

Merci pour votre et réponse et bonne chance.
MICHAUD Baptiste

baptistemichaud, 2024-03-18:
merci beaucoup 
bonne soirée 
73'