armel/uv-k5-firmware-custom-feat-F4HWN #33: Still transmit 1 sec after release of PTT

lapinos03, 2024-03-17:
This bug appears only in VFO mode, band F3 (144MHz), on TX_CHANNEL A.
Push PTT or the 1750Hz button, release it and the talkie continues to transmit during 1 sec approx. 
No problem in Memory mode, whatever band, whatever TX_CHANNEL. No problem in VFO mode on TX_CHANNEL B.

V2.3 loaded and tested on UV-K5 and UV-K5(8). I had no issue with V2.2.

armel, 2024-03-17:
I can't reproduce it. 
You're using Classic PTT or OnePush PTT (Menu 56/63 SetPtt) ?

Armel F4HWN.

armel, 2024-03-17:
By the way, what about STE (Menu 41/63) and RP STE (42/63) ?
STE is may be ON with RP STE set to 10*100ms...

Armel F4HWN.

lapinos03, 2024-03-17:
Wrong alert ! This was due to PTT-ID activated, inadvertently on both units. I put it off and now it's all good.
Sorry for the disturbance!

armel, 2024-03-17:
Ok, no problem 🤗

closed 2024-03-17 by armel