armel/uv-k5-firmware-custom-feat-F4HWN #35: V2.3 Modifs non prise en comptes

jcs33610, 2024-03-22:
Bonjour,
Je suis radioamateur, récent possesseur d'un UVK5(8) et nouveau dans ce domaine.
J'ai installé la V2.3 et me suis rendu compte que certaines modifications comme le changement de Narrow en Wide ou la modification du CTCSS n'étaient pas prises en compte ça a l'air de s'enregistrer mais si on change de canal et qu'on revient sur celui modifié la modif n'a pas été prise en compte.

Merci pour ton travail.
Amicalement
F4JNY Jean-Claude

armel, 2024-03-22:
Bonjour,

C'est normal. Vous n'utilisez pas correctement ce poste.
Si vous souhaitez que ces changements persistent, vous devez sauvegarder à nouveau le canal mémoire via le menu 15/63 ChSave. Sinon, le changement ne sera que temporaire. Et c'est logique.

73'

Armel F4HWN.

jcs33610, 2024-03-22:
Merci Armel je vois que j'ai encore beucoup de choses à apprendre ;)

73's
Jean-Claude F4JNY

closed 2024-03-22 by jcs33610