armel/uv-k5-firmware-custom-feat-F4HWN #36: Low power settings on UHF very low

oe3jtb, 2024-03-24:
I have version 2.3
since 2.2 this improvement was made
_improve default power settings level (Low ~125mW, Mid ~2W and High ~5W),_
at VHF this wokrs perfect, 
within menu 55 you can set low power L1-L5. The values are working very fine as displayed at VHF, at UHF even L5 is only 250mW.
measured with Bird 43, Diamond SX1000 and TINY SA Ultra, always the same results.
tnx
73 Alex OE3JTB

armel, 2024-03-25:
Hi Alex,

Thank your for your message.

I controlled again with my Diamond SWR Meter and my TinySA Ultra, on 435 MHz. L5 is a little bit lower on UHF than VHF, but it around 850~900mW. 

Could you please repeat the test with your TinySA? Could you measure again in UHF, L1, L2, L3, L4 and finally L5 and in this order. It could help me.

Here are my measures :

**VHF 145.000 MHz**

```
L1 2mW (for me this power is to low to be measure with TinySA Ultra...)
L2 118 mW
L3 264 mW
L4 470 mW
L5 937 mW

```
**UHF 435.000 MHz**

```
L1 35 mW
L2 70 mW
L3 180 mW
L4 314 mW
L5 650 mW
```

Not that with my Diamond SWRMeter, VHF or UHF is equal. 

73'

Armel F4HWN.

armel, 2024-03-25:
Another test. I tried with a TinySA Basic (so not the Ultra one)...

**VHF 145.000 MHz**

```
L1 1mW 
L2 100 mW
L3 249 mW
L4 445 mW
L5 885 mW
```

**UHF 435.000 MHz**

```
L1 46 mW
L2 130 mW
L3 290 mW
L4 516 mW
L5 818 mW
```

73'

Armel F4HWN.

oe3jtb, 2024-03-25:
Hi Armel
here you see, VHF is ok, L1-L5, UHF scary 
https://ibb.co/dpNTSYV
https://ibb.co/X5smzfw
https://ibb.co/fF9vyyf
https://ibb.co/nQf7g3B
I will try to reflash it with another firmware take measure and than reflash yours, Very intresting issue, as an UHF M and H is working ok
73 Alex, OE3JTB

oe3jtb, 2024-03-25:
So did it, flashed IJV 3.0 and EGZUMER 0.22, low power as expected 0,5W flashed to yours and it was the same result as before flashed the others

oe3jtb, 2024-03-25:
Hi Armel, is the readout with the k5prog, read calibration?

armel, 2024-03-25:
Hi Alex,

Yes, and just an extract between address 1ED0 and 1F30.

Armel F4HWN.

oe3jtb, 2024-03-26:
Here is mine Armel

32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff 
32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff
5f 5f 5f 69 69 69 77 87 89 ff ff ff ff ff ff ff
32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff
5a 5a 5a 64 64 64 82 82 82 ff ff ff ff ff ff ff
5a 5a 5a 64 64 64 89 7b 8a ff ff ff ff ff ff ff
32 32 32 64 64 64 8c 8c 8c ff ff ff ff ff ff ff  

tnx
73 Alex

armel, 2024-03-26:
Hi Alex,

So for VHF 
```
Me -> 5f 5f 5f 69 69 69 87 87 87 ff ff ff ff ff ff ff
You -> 5f 5f 5f 69 69 69 77 87 89 ff ff ff ff ff ff ff
```

And fo UHF
```
Me -> 5f 5f 5f 69 69 69 87 87 87 ff ff ff ff ff ff ff
You -> 5a 5a 5a 64 64 64 89 7b 8a ff ff ff ff ff ff ff
```

There are some small differences. But it didn't explain your measures in UHF.

Armel F4HWN.

oe3jtb, 2024-03-26:
Hmh, scary, in the next days I will receive a brand new one and I will try to flash version 2.3 at the first modification and will report.
If I flash IJV to that (affected) radio, the power is exactly what it should be in low 0,5W. But there is only one level , not 5 as in your firmware version. 
when you set the low level in your firmware are the same bytes modified in each L1-L5 level? And if yes wich onces? So perhaps I can flash IJV set low to 0,5 W and than flashing yours and set also to 0,5W and compare the UHF hex values? 
Or are they only for reference (init value) to a delta value in another register, so I cant compare? 
tnx
73 Alex

armel, 2024-03-26:
Ok so, here is an example. For UHF band, as I said, I have this :

5f 5f 5f 69 69 69 87 87 87 ff ff ff ff ff ff ff

5f is for low power, 69 is for mid power and 87 is for high power.

5f in hexadecimal is equal to 95 in decimal. 
69 in hexadecimal is equal to 105 in decimal.
87 in hexadecimal is equal to 135 in decimal.

L1  is compute like this : (95 * 4) / 25 = 15
L2  is compute like this : (95 * 4) / 19 = 20
L3  is compute like this : (95 * 4) / 13 = 29
L4  is compute like this : (95 * 4) / 10 = 38
L5  is compute like this : (95 * 4) / 7 = 54

Mid  is compute like this : (105 * 3) / 4 = 78

High  is compute like this : (135 + 30) = 165

In attachement, my measure on UHF (435.000 MHz) with L5 and TinySA or Diamond SX40. 

Armel F4HWN.

[image]
[image]

armel, 2024-03-26:
By the way, as I said many time : always backup your eeprom (for example with k5prog) **BEFORE** install exotic firmware ☺️ You could restore your pocket as it was out of the box if necessary. 

Armel F4HWN.

oe3jtb, 2024-03-26:
yes backup should be in the other qth in Vienna, will search for it next week :-)

oe3jtb, 2024-03-26:
Could it be the problem here

- You -> 5a 5a 5a 64 64 64 **89 7b 8a** ff ff ff ff ff ff ff
so there are not the same values 3 times?

oe3jtb, 2024-03-26:
So with your help Armel I made a workaround
mine  5a 5a 5a 64 64 64 89 7b 8a ff ff ff ff ff ff ff
I changed with IJV firmware (there you can modify L/M/H power by menu and the hex value you see in display)
to
 **5f 5f 5f** 64 64 64 89 7b 8a ff ff ff ff ff ff ff
and now it works, not measured, but with L4 I can open a repeater 60km away, so power must be better :-)
Tnx for your help and advice, perhaps this help others too
73 Alex, OE3JTB

armel, 2024-03-27:
Ok Alex, you're welcome ;)

Armel F4HWN.

oe3jtb, 2024-03-27:
So last findings with the Tiny SA Ultra, power is much better, the deviation is bwtween a quarter and on-third lower as the setup in the menu, perhaps I will play with setting a higher value with IJV and than flashing back to your FW, to reach nearly the L1/L5 values. But at all it works fine also now
73 Alex, OE3JTB

closed 2024-03-27 by oe3jtb