armel/uv-k5-firmware-custom-feat-F4HWN #37: No TX translation with external microphone in monitoring mode.

FAH5502, 2024-03-24:
I have a probleme with the external microphone PTT in monitoring mode or squelch at 0.
the PTT button not working as intended.
but the PTT on the radio work well in monitoring mode.
can you fix this problem.
Best regards.
FAH5502.
Marc

armel, 2024-03-24:
Marc, sorry, it's not a firmware issue. 
It's a microphone (and hardware) issue.

Take a look here :

[image]

Armel F4HWN.

oe3jtb, 2024-03-24:
or you modify the radio itself, its easy to do https://github.com/ludwich66/Quansheng_UV-K5_Wiki/wiki/Hardware_Mods#5-fix-power-failure-with-external-handset-warming-speaker-with-dc-on-output-or-unwanted-busy-channel-lockout-effect-on-external-ptt

closed 2024-04-13 by armel