armel/uv-k5-firmware-custom-feat-F4HWN #38: STE doesn't seem to work when transmitting to Bao UV5R

terminalvelocity23, 2024-03-27:
v2.3, STE on on Quansheng UV-5R Plus, squelch noise heard at the end of transmission when receiving the signal on the Baofeng UV5R regardless of the STE settings on both stations. On a SDR receiver the STE tone being transmitted is heard, but it's shorter and lower in pitch than the corresponding tone of the UV5R. Maybe this is what causes the interoperability issues.
STE tone recordings in the attached file.

On reception, STE works flawlessly.

[ste tones.zip](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/files/14775792/ste.tones.zip)

armel, 2024-03-28:
Hi,

Thank you.
I will take a look on this.

73'

Armel F4HWN.

K5Quan, 2024-03-28:
Félicitations. Dans votre firmware, la fonction STE ne fonctionne pas correctement à la fin du transfert. S’il vous plaît, corrigez-le

https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/164750210/8b3bf597-f129-4742-a723-c58ef328a388