armel/uv-k5-firmware-custom-feat-F4HWN #40: S-meter wrong.

GonzoG, 2024-03-30:
S-meter calibration is still wrong. It's set to HF standard (S9 = -73dB), not VHF/UHF standard (S9 = -93dB, S0 = -140).
IARU-R1 VHF Handbook v.9.01, 2021:
_"Standardisation of S-meter readings
• One S-unit corresponds to a signal level difference of 6 dB,
• On the bands below 30 MHz a meter deviation of S-9 corresponds to an available power of -73
dBm from a continuous wave signal generator connected to the receiver input terminals,
• **On the bands above 30 MHz this available power shall be -93 dBm**"_

I know it can be changed with Chirp, but it should be correct in fw.

armel, 2024-03-30:
First, hello and thank you for you message...
Second, you're wrong. My S-meter is set to the VHF/UHF standard (S9 = -93dB, S0 = -140).
My S-meter table is this one :

```
S9 + 40 dB	-53 dBm
S9 + 30 dB	-63 dBm
S9 + 20 dB	-73 dBm	
S9 + 10 dB	-83 dBm

S9  -93 dBm	
S8  -99 dBm
S7 -105 dBm
S6 -111 dBm
S5 -117 dBm
S4 -123 dBm
S3 -129 dBm
S2 -135 dBm
S1 -141 dBm
```

If it's not the case on your K5 with my firmware, it's probably because your calibration eeprom is corrupt or was modified by an exotic firmware. That's not my problem and I can't control it. My firmware never overwrite calibration datas...
[image]

Take a look at the photo in attachement : -93dbm on the TinySA output = -91dbm on the K5 (so around S9).

Hope this help.

Armel F4HWN.

oe3jtb, 2024-04-02:
I also can confirm Armel, S-Meter is working very accurate

dg9vh, 2024-04-02:
Can also CFM, works very accurate.

armel, 2024-04-07:
No reply from the original author.

closed 2024-04-07 by armel