armel/uv-k5-firmware-custom-feat-F4HWN #41: calibration power chirp

f4axk, 2024-03-31:
Bonjour Armel

Depuis la version 2.4 on ne peut plus géré la calibration des puissances dans chirp (calibration power).
Très utile pour mon utilisation perso.

Bonne journée
Laurent

armel, 2024-03-31:
Bonjour Laurent,

Et oui, je n'en ai volontairement pas parlé, lors de la présentation de la
version 2.4.

La raison : beaucoup trop d'utilisateurs de K5 font tout et souvent
n'importe quoi. Ils testent des firmwares sans comprendre ce qu'ils font.
Et notamment le firmware IJV. Et ce firmware, comme d'autres, modifie en
profondeur les données de calibration. Ce que mon firmware ne fait jamais.
Donc, après avoir joué avec IJV, comme c'est une usine, ils reviennent sur
mon firmware et là, c'est le drame. La calibration des puissances n'est
plus la bonne. Et je reçois des messages me disant, par exemple, que même
en Low5, le poste ne sort que quelques milli watts de puissance... Bien
évidemment, la majorité n'a jamais pensé à faire un backup de l'eeprom
AVANT.

Franchement, ces firmwares close source qui en plus modifient l'eeprom,
c'est la plaie.

J'ai donc fait le choix de coder en dur, sur la version 2.4, l'étalement
des puissances (évidemment, sans toucher à l'eeprom).

Si ça te dérange vraiment, je peux tenter de te compiler un firmware avec
une gestion des puissances comme avant. En espérant que ça loge en mémoire,
ce que je n'ai pas vérifié.

73'

Armel F4HWN.

-- 

Armel FauveauDIRECTEUR ASSOCIÉ [image: Logo]

MAIL : ***@***.***

INDICATIF : F4HWN

[image: Icon website] <https://www.globalis-ms.com/> [image: Icon LinkedIn]
<https://www.linkedin.com/company/globalis-media-systems/> [image: Icon
Twitter] <https://twitter.com/GLOBALISms>

ACTUALITÉS

[image: Icon plus]
<https://www.globalis-ms.com/actualites/certification-label-scorefact/>

Globalis obtient la certification ScoreFact
<https://www.globalis-ms.com/actualites/certification-label-scorefact/>

f4axk, 2024-03-31:
OK  Armel 
j'avais bien vue les modifications dans radio.c ,j'ai refais une compile moi même donc pas de problème pour moi.

Le problème de l'avoir coder en dur me donne des puissances différentes sur mes deux postes.
Poste 1:  low 500mW ok, mid 1,5W, high 4W
Poste 2:  low 1W pour avoir 500mW, mid 2W, high 5W
Mes 2 postes on leur calibration d'origine, vas donc comprendre.

Voila pourquoi j'ai remodifier le code dans radio.c et re-compilé.
Pour ma part je règle mes poste 500mW, 2W, 4W.

il faut espérer que d'autre OMs n'est pas le problème.

Laurent F4AXK

armel, 2024-03-31:
Ok Laurent.

C'est un sujet super intéressant.

Alors, en premier lieu, dans la branche _feature_update_ et dans le répertoire _archive_, tu trouveras 2 versions de firmware 2.4 : 

```
archive/f4hwn.fixed.packed.v2.4.bin
archive/f4hwn.unfixed.packed.v2.4.bin
```

La première correspond donc à la version de production avec les puissances fixées.
La seconde fonctionne comme avant et se base sur les données de calibration (donc modifiable avec Chirp).

En complément, toujours dans la branche _feature_update_, j'ai ajouté une directive de compilation dans le Makefile. A savoir ENABLE_FEAT_F4HWN_FIXED_PWR   que l'on peut donc initialisé à 1 (pour fixer les puissances) ou 0 (pour utiliser les données de calibration). 

En complément, pour avoir fait pas mal de mesures, d'un pocket à l'autre (j'en ai 5) : aucun poste ne sort les mêmes puissances. Il y a énormément de disparité d'un modèle à l'autre. J'ai même un K5 (8) qui, sorti du carton, frôle les 7 watts en émission en HIGH sur la VHF !!! Le pire, c'est qu'il se met très vite en sécurité (l'écran commence à se brouiller et même s'inverser, puis s'éteint). 

Le problème reste tout de même le même : tous ces firmwares exotiques et close source, comme IJV, qui modifient pas mal de choses en profondeur. Comme dit, c'est la plaie !!!

73'

Armel F4HWN.

f4axk, 2024-04-01:
Bonjour Armel

Merci pour les fichiers et l'option dans le Makefile (ENABLE_FEAT_F4HWN_FIXED_PWR), beaucoup plus simple pour compiler.

Bonne journée 
73s
Laurent F4AXK

armel, 2024-04-01:
Bonsoir Laurent,

Avec plaisir. Je cloture cette issue. 

73'

Armel F4HWN.

closed 2024-04-01 by armel