armel/uv-k5-firmware-custom-feat-F4HWN #43: Weird scanner behavior, the radio gets stuck in the RX mode on exit

terminalvelocity23, 2024-04-02:
Hi Armel,
v. 2.4, K5(8) and UV-5R Plus both show the same behavior. After exiting the frequency scanner mode, if the radio was receiving a signal at that time, it gets stuck in RX mode apparently - the green LED is on, and after the power save timeout it starts clicking repeatedly. The frequency of the clicks is depends on the power save ratio setting, the higher the setting, the lower the frequency. If power save is disabled, there's no clicking, but the RX LED remains lit indefinitely. To stop this, you need to switch VFO, change frequency setting or press the TX button, there's a single click, the LED switches off and the clicking stops.

This only happens in XB or MO RX modes, DWR and DW are unaffected.

After resetting the radio to default settings there's no more clicks, but the LED stays on nevertheless. I've set the station with my preferred settings via CHIRP and the clicking is back.

See the attached CHIRP configuration file and the video.

73!

[Quansheng_UV-K5 (egzumer + f4hwn)_20240402 reset.zip](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/files/14837115/Quansheng_UV-K5.egzumer.%2B.f4hwn._20240402.reset.zip)

https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/145500871/f9a2c87d-6814-4642-8f83-a1a29024b1f7

armel, 2024-04-02:
Hi,

Okay, thank you. I can reproduce the bug !

Well, it's quite a specific procedure. I'd never thought of doing such a thing : switch from scanRange to the spectrum analyzer en go back.

I will try to take a look, but I'm not sure to spend a lot of time on it. 

73'

Armel F4HWN.

terminalvelocity23, 2024-04-02:
Hi Armel,

It's not neccessary to go from the ScanRange to spectrum analyzer to produce the bug, you only need three things to happen - RX mode MO or XB, spectrum analyzer on and receiving something and pressing exit at that time.
I'm using the ScanRange function to define the spectrum analyzer bandwidth most often, it's very handy.

Here's the video of it.
Anyway, it's not a crippling bug so no worries. Thank you again for your work!

https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/145500871/954c786f-2f85-4126-b7bd-4632cd92b4f3

armel, 2024-04-02:
Ok,

Could you try the firmware in attachement ?
Be aware, you must unzip it before flashing it, because GitHub issues don't accept .bin extension.

Tell me,

[f4hwn.packed.bin.zip](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/files/14839665/f4hwn.packed.bin.zip)

Armel F4HWN.

terminalvelocity23, 2024-04-02:
Sure I can.

Flashed it to UV5R Plus, can't reproduce the bug in question anymore no matter what I'm trying.

armel, 2024-04-02:
Thank you for this test.
So it's a good news. I think it's fixed.
I will close this issue if it's ok now.  

Armel F4HWN.

terminalvelocity23, 2024-04-02:
Yeah, it works fine now, thank you. Tested it with K5(8) also, no issues.

armel, 2024-04-02:
Fixed in v2.5

closed 2024-04-02 by armel

terminalvelocity23, 2024-04-02:
That was super quick! Thanks again.