armel/uv-k5-firmware-custom-feat-F4HWN #44: Expanding the frequency range below 18MHz

zylka5, 2024-04-04:
I kindly request to extend the receiving frequency range limit below 18MHz. Within the 17MHz band, there are several stations that are very well received by the Quansheng. Recently, I also received communications on 16MHz. This simple change would cost 0 bytes of memory and would bring a lot of joy :D

armel, 2024-04-04:
Hi,

With or without Si4732 mod ?
According to the datasheet, BK4819 cover only 18 MHz ~ 660 MHz and 840 MHz ~1300 MHz.

Armel F4HWN.

armel, 2024-04-04:
Hello,

My firmware is open source, of course :)
The firmware source code is available on this [repo](https://github.com/armel/uv-k5-firmware-custom) ;)

Regards,

Armel F4HWN.

zylka5, 2024-04-09:
I actually overlooked that, now I have a problem with compilation. It seems to be successful, but I don't see the resulting .bin file. Where can I find it?
[image]

armel, 2024-04-09:
As usual, in _compiled-firmware_ directory.

Armel F4HWN

zylka5, 2024-04-11:
Strange issue, the compiled-firmware folder is created but it only contains a single file named firmware.ld. The .bin files are missing. What could be the problem?

armel, 2024-04-11:
I don't know. I'm under MacOS.

Armel F4HWN.

spamsool, 2024-04-11:
Same problem here, I built with docker and only firmware.ld in present compiled-firmware directory
Missing firmware file.

armel, 2024-04-11:
Maye be take a look at _compile-with-docker.bat_ windows script.
I saw this code

```
@echo off
docker build -t uvk5 .
docker run --rm -v %CD%\compiled-firmware:/app/compiled-firmware uvk5 /bin/bash -c "cd /app && make clean && make && cp firmware* compiled-firmware/"
pause
```

Change it by this one, and test again...

```
@echo off
docker build -t uvk5 .
docker run --rm -v %CD%\compiled-firmware:/app/compiled-firmware uvk5 /bin/bash -c "cd /app && make clean && make && cp f4hwn* compiled-firmware/"
pause
```

Note I change firmware* by f4hwn* ...

Armel F4HWN.

spamsool, 2024-04-11:
Thank you @armel 
I can confirm that your sugestion change firmware* by f4hwn* in compile-with-docker.bat solved the building issue. I was able to succesfully build f4hwn.packed.bin with my small change (reverse battery symbol) and the file is still 4 bytes smaller then the one on the github directory. Thanks.

I think @zylka5 can now try to build his version

zylka5, 2024-04-13:
It's working now! I used the new v2.5 source code and didn't have to change anything. Thank you for your help!

armel, 2024-04-13:
Ok, nice.
It's because I patched the _compile-with-docker.bat_ that you're using under Windows.
I close this issue.

Armel F4HWN.

closed 2024-04-13 by armel