armel/uv-k5-firmware-custom-feat-F4HWN #45: Backlight timeout and buttons shortcuts

xes, 2024-04-05:
If possible I would ask about some little enhancement.
The first is about backlight timeout setting. Would it be possible to add more values betwen 20 seconds and 1 minute?
For example 30, 40 secs...

About shortcuts, it would be great to have some more functions available to be assigned to F1/F2/M  for example spectrum and squelch level.

armel, 2024-04-05:
Hi,

Thank you for your message.
It's possible, off course, but the question is : why ? Have you got a typical use case ? Note that, you could use F + 8 to quickly switch backlight ON or OFF, on demand.

Armel F4HWN.

armel, 2024-04-05:
Ok. I looked at the code. 

In the next version, you'll be able to set the value of BackLt to ON or OFF or for a duration ranging from 5 seconds to 5 minutes... by steps of 5 seconds. For example, you could choose 03min 15sec.

Enjoy.

Armel F4HWN.

zylka5, 2024-04-05:
Since there is a smooth brightening when turning on, it could be added at any time :D

armel, 2024-04-11:
Fixed in v2.5

closed 2024-04-11 by armel