armel/uv-k5-firmware-custom-feat-F4HWN #46: Battery percentage differences

LHunti, 2024-04-05:
Hi Armel,
I experienced an interesting phenomenon regarding the battery percentage of one of my radios and I could not find an explanation for it. Apparently, the calibration values are fine, but still the 2 radios show a huge difference. Have you ever encountered such a phenomenon? Thank you in advance for your reply, as well as this firmware that you have improved in a super direction.
[image]

LHunti, 2024-04-05:
Yes, of course, I performed the calibration and the voltage value written by the radio is real, but the battery symbol and the percentage value do not reflect reality.  I just don't understand how this is possible?  It's true that I haven't reviewed the source code that applies to this, but the values stored in the eeprom seem relatively good, and if I understand correctly, the percentage values and the symbol also work from these values, right?

armel, 2024-04-05:
Humm, ok.

It's like your battery on K5(8) is a 1600mAh and your battery on K5 is a 2200mAh...
Take a look to the Voltage to Percentage Table :

```
const uint16_t Voltage2PercentageTable[][7][2] = {
	[BATTERY_TYPE_1600_MAH] = {
		{828, 100},
		{814, 97 },
		{760, 25 },
		{729, 6  },
		{630, 0  },
		{0,   0  },
		{0,   0  },
	},

	[BATTERY_TYPE_2200_MAH] = {
		{832, 100},
		{813, 95 },
		{740, 60 },
		{707, 21 },
		{682, 5  },
		{630, 0  },
		{0,   0  },
	},
};
```

Regards,

Armel F4HWN.

LHunti, 2024-04-06:
Sorry for wasting time, because it seems I started this post in a hurry, because this phenomenon can be observed with both radios.  If I set the battery type to 1600mA, the battery symbol on the UV-5R will also be empty.  According to them, would there be such a big difference in the discharge curve between the two battery types?

armel, 2024-04-06:
Hi,

No problem :) 
I confirm, there is a big difference in the discharge curve between 1600mAh and 2200mAh batteries.

Armel F4HWN.

closed 2024-04-07 by LHunti