armel/uv-k5-firmware-custom-feat-F4HWN #47: Not supported by CHIRP ?

z34pec, 2024-04-12:
Hello.
When I try to acccess UVK5 with latest firmware 2.5 I receive message that firmwaer is not supported.
Also happened version before of chirp but this time it was IJV 2.7.

[image]

z34pec, 2024-04-12:
I think find a solution.
I think this module should be added to your existing files.

https://chirpmyradio.com/issues/11270

f1ixx, 2024-04-12:
Hello
you can also read [driver here](https://github.com/armel/uv-k5-chirp-driver)

73's
Boris

armel, 2024-04-12:
Hi,

My firmware as many others need a dedicated driver. Take a look at [this](https://github.com/armel/uv-k5-chirp-driver) :) 

Regards,

Armel F4HWN

closed 2024-04-12 by armel

z34pec, 2024-04-13:
Thanks. I think this driver should be added in FW package.