armel/uv-k5-firmware-custom-feat-F4HWN #48: Delay on RX after power on (AM band)

lapinos03, 2024-04-14:
Hi !
You solved the delay on RX after power on... on the VHF. It actually works, I tested it. But not on AM band. I have a channel on the ATIS Clermont-Ferrand airport 136.400 (TX continuously all day long) and when I power on, whatever PonMessage value, it waits for about 7s to activate the RX.

armel, 2024-04-14:
Is this bug also present if you are in RxMode MAIN RX DUAL RX or DUAL RX RESPOND or only in MAIN ONLY? 

Armel F4HWN.

lapinos03, 2024-04-14:
VFO-A: VHF (memory or value), VFO-B: airband AM (memory)
Power-on : 
MO : ok
DW : bug
DWR : bug
Cross : ok

When I cycle through out the modes with long press on M key, with VFO-A as main and VFO-B as secondary, there is also a delay for the second VFO : 
MO : not applicable
DW : 4s
DWR : 4s
Cross : 0s

When VFO-B as main : 
MO : 0s
DW : 0s
DWR : 0s
Cross : not applicable

/Pascal

armel, 2024-04-14:
Ok Pascal, thank you.
Sorry, but I'm not able to reproduce this problem.
Take a look at the video in attachement.

https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/127889/d54830d5-856f-4037-a9a8-2c67d4641ca7

Maybe I'm doing it wrong?

Armel F4HWN

lapinos03, 2024-04-15:
Lucky you !
Your test signal is strong. On my side I receive the airport à -92dBm.

I made 2 videos (which I will post from my mobile phone in next post). In the second one, I switch from A to B and vice versa many times, you will see the delay. I made several tests with different POnMessage and channel B on MR or VFO.

lapinos03, 2024-04-15:
Damned ! Video must be less than 10MB...

lapinos03, 2024-04-15:
https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/891923/beea2dbd-31b5-4372-9363-64d6cb9afe16

lapinos03, 2024-04-15:
Uploading 20240415_204329_red.mp4…

lapinos03, 2024-04-15:
https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/891923/25813223-8a8e-4828-bc6a-7fff7c951d23

lapinos03, 2024-04-15:
I got it !
When the AM channel is in WIDE mode then there is the issue. In NARROW mode, no more problem, it starts to receive the AM right away after power-on, in both MR and VFO mode.

So, to sum up : 
1. MO : AM/Wide or AM/Narrow -> ok
2. DW / DWR : AM/Wide -> bad (whatever VFO A or B)
3. DW / DWR : AM/Narrow -> ok

In your video, you are in MO mode WIDE bandwidth for the AM.

armel, 2024-04-15:
Hmmm ok, that's really strange. 

So the solution is probably to switch to narrow mode. I can't really see if it's a bug or how to improve it. I rarely use AM listening and I'm not an expert of this modulation. 

Armel F4HWN

closed 2024-05-05 by armel