armel/uv-k5-firmware-custom-feat-F4HWN #50: Amélioration Mode "Main Only"

magellan-13016, 2024-04-14:
Bonjour Armel,

Felicitations pour le dernier firmware 2.5 auquel je ne m'attendais pas. J'ai vu que beaucoup améliorent leur K5 avec une nouvelle EPROM alors peut-être qu'on échappera à la suppression du Broadcast FM :-) .

Je me permets juste une suggestion pour le mode Main Only : Tu affiche le VFO actif an bas à droite en inversé ce qui pour moi altère grandement la lisibilité. C'est le seul élément affiché en mode inversé dans ce mode et j'aimerai bien qu'il soit affiché comme le reste.

Maintenant, te connaissant, je pense que tu vas me faire la démonstration que c'est très bien ainsi et que ça ne changera pas... mais au moins j'aurais essayé ! ;-)

73, David (F4BPP).

magellan-13016, 2024-04-14:
Je me doutais bien que j'allais garder ce truc illisible en bas en droite... je l'ai même écrit noir sur blanc avant ta réponse ! :-) Et c'est là que j'apprécierai un firmware à compiler sois-même avec des petits "#" à mettre ou à enlever selon ses préférences et un petit tuto pour le faire... mais bon, comme on dit, à cheval donné, on regarde pas les dents ! Hi ! 

J'entends bien que c'est le seul truc sur lequel tu peux agir sur un écran monochrome niveau UI, la question est : "Est-ce vraiment utile ?" Pour l'astronomie, le reverse total, je comprends mais le mix des deux, je ne suis vraiment pas fan du tout... 

Mais comme d'habitude, je vais me faire une raison.

Par contre, qu'est-ce que j'ai aimé le recadrage sur l'émission en HF ! Enfin quelqu'un qui ose dire les choses et fait de la prévention sur le spectre électromagnétique ! C'est ta vidéo où tu explique que le K5 est dans les clous niveau harmoniques que j'ai pris la décision d'acheter le mien ! C'est à se demander si au niveau de la licence, il ne devrait pas y avoir plus de sensibilisation à une émission propre.

73, David.

armel, 2024-04-14:
Tout à fait, tu as bien raison pour la sensibilisation concernant les notions de pureté spectrale. Ce serait une excellente idée. Peut-être que cela viendra, à l'occasion d'un dépoussiérage du programme abordé... on peut rêver :)

Et effectivement, pas d'évolution avec le K5-99, coté augmentation de l'espace de programmation. 

J'ai communiqué à Quansheng les références d'un Cortex M0 qui pourrait, me semble t-il, faire l'affaire. Mais la balle est dans leur camp. Je suis toujours en contact avec eux.

73'

Armel F4HWN.

closed 2024-04-15 by armel