armel/uv-k5-firmware-custom-feat-F4HWN #52: Screen garbling at 2m TX with full power

terminalvelocity23, 2024-04-15:
Hi Armel!

v 2.5 (and earlier versions as far as I can tell) display the following  behavior - when transmitting at 2m, the screen gets garbled, turned upside down or blanked out if you don't hold the radio in your hand and use One Push PTT. It reverts back to normal when you take the radio back in your hand in some cases, and always reverts back when you turn the transmission off. If you hold the radio in your hand, there's no problem. The exact frequencies depend on the antenna in question, but with the stock antenna I can reproduce it between 145 and 149 MHz.
Only transmitting at high power causes this effect, mid and low power modes doesn't seem to do that.
This doesn't happen with the stock firmware v.2.6 at the same frequencies. 
Both of my radios, UV5R Plus and K5(8) show the same behavior.

I've tried reflashing the radios and resetting the settings, and this didn't help. Resetting the calibration to stock values didn't help, either.

Video's too big to attach it here, so here's [the cloud link](https://disk.yandex.ru/i/ceLvCueicicQ6w). And my CHIRP config file in the zip archive.

Thanks for your work!

[Quansheng_UV-K5 (egzumer + f4hwn)_20240415_new_squelch.zip](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/files/14984307/Quansheng_UV-K5.egzumer.%2B.f4hwn._20240415_new_squelch.zip)

armel, 2024-04-15:
Hi,

Have you tested IJV or an another firmware? Some calibration data leads me to that it's the case. 

[image]

Try to reduce these values, on band 3, using Chirp (and my Driver of course)... 
Replace high values lower (119), center (143), upper (137) by lower (130), center (130) and upper (130).

143 is really too high (I never saw this value before...).

Armel F4HWN.

terminalvelocity23, 2024-04-15:
Tried that, didn't help. Flashed the stock calibration with k5_prog, didn't help too. 
Even setting it as low as 110 doesn't appear to do anything.

armel, 2024-04-15:
My stock calibration high value is 135.
My firmware increase the default high value and add 30, so when the transceiver start, this value is set to 165 (135 + 30 for me). 
All my transceivers work like a charm with this correction. And high power is over 5 watts (around 5,5 watts)...

Your transceiver (and the BK4819 chip in your transceiver) is probably not able to support this correction. Why, I don't know. But there are sometimes some diff between 2 BK4819... 

So, if it works with stock firmware (as you said) and if your default stock calibration high value is, for example, 120... set it to 90 (90 + 30 = 120).

Armel F4HWN.

terminalvelocity23, 2024-04-16:
Still garbling at 90, although somewhat less. I flashed your firmware to a new K5(8) right out of the box, and it worked fine with the stock calibration settings (123, 137, 123). That's interesting. Maybe I did something to my radios which I shouldn't have done.

armel, 2024-04-16:
Ok, it's very strange. Note that, you're the first to bring this problem to my attention. The effect on your video is quite impressive !!! I never saw that before. Is the transceiver became hot ? 

Armel F4HWN.

terminalvelocity23, 2024-04-16:
Not at all, it's immediately apparent. 
For comparison, I've blasted full power with both IJV and stock firmwares for 5 minutes straight at the same frequencies, the transceivers got somewhat hot, but no screen glitches at all.

armel, 2024-04-16:
The problem is not visible on UHF ?

Armel F4HWN.

armel, 2024-04-16:
Okay, I can reproduce your problem on VHF only.

But my analysis is simple: it's just a problem of antenna and SWR. 

With the standard antenna, if I put the transceiver on a table during a transmission (with the OnePush function...), the problem appears. But it's important to understand that, on a pocket, antennas are designed and optimized to work in synergy with the walkie-talkie chassis and, crucially and impactingly, **with the user's body, which acts as a counterweight**.

By the way, if I hold the pocket in my hand, normally, or use an alternative antenna with a good counterweight, the problem disappears. 

Armel F4HWN.

terminalvelocity23, 2024-04-16:
That's true, the portable transceivers are supposed to be held in hand. Still, no other firmware displays the same issue - I've tested them in same conditions, the output power looks the same on the receiving side, the position is the same, and with the same antenna. 
I suspect that some arcane stuff about timings is to blame for this.

Oh well, on the other hand it's handy for finding the resonant interval of your VHF antenna :)

armel, 2024-04-16:
Hello,

Ah ah, agree, that's true about finding the resonant interval of your VHF antenna :)

By the way, I noted that, when the problem appears, with the pocket on the table, the antenna base (near the connector) is really hot. Again, according to me, it's just a SWR issue. And it's normal.

About others firmwares, the original firmware, IJV or my firmware use differents power settings. So, it's normal that there are some differences. 

My firmware use the same code for mid power. And the problem is not visible. Why ? Because SWR is less important with a lower power. And it's normal too. The percent of power return in the pocket is less. 

Last but not least, no problem with UHF. Why ? Because antenna is more adapted. Quarter wave antenna, etc. 

The original firmware don't output 5 watts (or more) with high power. About IJV, I don't know, it's a closed source firmware so I don't use it. I only know that IJV change many things on calibration and it's really a problem for others firmwares developpers and users who don't know how backup and restore calibration datas.

Finally, the K5 is a pocket and need to be us as is :) If it's a problem, use mid power (with a less power return) or decrease high power calibration settings or... use another firmware :) 

Armel F4HWN.

terminalvelocity23, 2024-04-16:
No worries, the fact that you've been able to replicate it at least means that I haven't fried my radios.
Cheers!

terminalvelocity23, 2024-04-16:
Just some current draw numbers FYI

W @ 8,3V (RA-771)
144                  433
k5
hi 8.8		9.1
med 5.5		5.6

uv5r plus
hi 9.7		7.6
med 6.4		6.3
lo5 4.5		4.6

closed 2024-04-27 by armel