armel/uv-k5-firmware-custom-feat-F4HWN #55: Stuttering audio on NFM mode

dadanielat, 2024-04-20:
When received audio is reaching a certain loudness level, stuttering occurs when channel is set to NFM mode. Also happens with stations using legit devices that only support NFM. Seems to be a bug of egzumer 0.22, because in 0.21 it was ok.
I would suggest a quick fix of only sending in NFM and always receive in FM mode.
Was reported by many users in https://github.com/egzumer/uv-k5-firmware-custom/issues/467

armel, 2024-04-20:
Ok, I've never encountered this problem.

Have you take the time to read this response from [Egzumer](https://github.com/egzumer/uv-k5-firmware-custom/issues/467#issuecomment-1937873361) ?

About your solution, only sending in Narrow FM and always receive in Wide FM mode, sorry but this is nonsense. 

The more simple is, if you receive Wide FM with radio set to Narrow, and if you've got this problem, simply switch from Narrow to Wide (using an action "SWITCH WIDE NARROW" on F1 short, for example). 

Armel F4HWN.

dadanielat, 2024-04-20:
I have read his comment, but I can assure you that I'm not receiving Wide FM because the sending device is only capable of NFM. Of course I can solve the receiving problem with manually switching to FM, but when I TX the other device receives distorted audio because it supports only NFM. I would have to switch back again every time before sending, so this cannot be the solution.

armel, 2024-04-20:
Sorry again, but I would't update K5 firmware to only sending in Narrow FM and always receive in Wide FM mode, because of some legit devices that only support Narrow FM... Transmit or received in NFM or WFM is a standard fonction of a transceiver. Why K5 needs to adapt it to others ?

By the way, I made many tests with a parrot on devices who uses Narrow FM or Wide FM, and there is no problem. 
So these legit devices you're talking about are may be defective? Out of curiosity, what are they ? Which models ?

Armel F4HWN.

dadanielat, 2024-04-21:
It seems there is a misunderstanding: Of course my proposed solution should only be effective when NFM mode is enabled for the channel. If NFM channel then TX NFM and RX FM. If FM channel then TX FM and RX FM.

ON7MHZ, 2024-04-23:
Same issue here , already for a long time ( I think since egzumer v13 or earlier) . 
I have this when i'm using my TM-733 in transponder mode with the quansheng.
Always though this handheld could not deal with very strong signals (although my transponder TX is only 5w)
Strange thing is that I tried IJV a few days ago, and there I had no problem with stuttering audio. 
I will record a video as soon as i can . 
PS : I really like the RX/TX timer on v26 !! 
73

armel, 2024-04-23:
Hello,

The solution proposed above is unreasonable. And I couldn't reproduce this issue. But if anyone can tell me **exactly** how to correct the problem and dealing with the BK4819 about this stuttering audio issue, _without degrading the transceiver's operation_, I can handle it. 

The IJV firmware is closed-source firmware, so maybe it changed something somewhere, but no one will ever know what... 

Regards,

Armel F4HWN.

ON7MHZ, 2024-04-23:
A way to reproduce might be a station with strong audio.
Like I said, I have it when using my transceiver as crossband transponder. As you can see in the vid the stuttering is not on the (unfortunately not open) IJV firmware ( left) . 

https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/154373720/fc48af82-2b0e-4abd-821c-c6f649eed8ec

lapinos03, 2024-04-28:
> When received audio is reaching a certain loudness level, stuttering occurs when channel is set to NFM mode. Also happens with stations using legit devices that only support NFM. Seems to be a bug of egzumer 0.22, because in 0.21 it was ok. I would suggest a quick fix of only sending in NFM and always receive in FM mode. Was reported by many users in [egzumer/uv-k5-firmware-custom#467](https://github.com/egzumer/uv-k5-firmware-custom/issues/467)

I too have encountered this issue. A guy with loud voice through a VHF relay while my receiver was in Narrow mode. Had to switch to WIDE bandwidth.

By the way, what is the bandwidth of narrow mode ? 12.5KHz ? 6.25KHz ?

armel, 2024-04-28:
Bandwidth of narrow mode is 12.5 kHz.

ON7MHZ, 2024-04-29:
> I too have encountered this issue. A guy with loud voice through a VHF relay while my receiver was in Narrow mode. Had to switch to WIDE bandwidth.
>
> By the way, what is the bandwidth of narrow mode [...]

As you can see on the display (right quansheng) the setting was already Wide FM here