armel/uv-k5-firmware-custom-feat-F4HWN #56: Ajout de fonctionnalité : TOT par fréquence en mémoire

fifisoft57, 2024-04-23:
bonjour

je voulais savoir si l'option TOT peut être active juste sur certaine mémoire en fréquence ,par exemple un relais qui est en mémoire et on active le TOT et une fréquence PMR en mémoire sans TOT ??

Merci

Cdt

Philippe

armel, 2024-04-23:
Bonjour Philippe,

Non, ce n'est pas possible en l'état. Le TOT est global. Il est appliqué à l'ensemble des canaux (et également au VFO). Maintenant, c'est plutôt une bonne chose. Ca évite les risques de passer en émission sans interruption, par inadvertence. C'est encore plus le cas avec l'option SetPTT OnePush. Une erreur de manip est vite arrivée. Et ca évitera de cramer votre poste :)

Armel F4HWN.

fifisoft57, 2024-04-23:
ok merci pour le retour, je comprend

armel, 2024-04-23:
Pas de problème. Je cloture :)

closed 2024-04-23 by armel