armel/uv-k5-firmware-custom-feat-F4HWN #59: Gestion mode OnePush - Suggestion

F6EGK, 2024-04-24:
Bonjour Armel,

Il s'agit ici d'une comparaison dans la gestion du mode "OnePush" par rapport à ce qui est implémenté dans OpenGD77. Un dessin valant mieux qu'un long discours, le chronogramme joint illustre cette comparaison. Ce que je relève d'un point de vue utilisateur, c'est que l'on a tendance à vouloir anticiper le passage de la configuration émission vers réception en appuyant sur le bouton PTT. Dans le cas de l'existant, l'UV-K5 repasse immédiatement en réception. Dans l'approche OpenGD77, le passage en réception n'intervient qu'au relachement du bouton PTT. 

[image]

Cela peut sembler anodin, mais on gagne en confort avec le même résultat final. C'est juste une suggestion. C'est peut-être simple à mettre en oeuvre, d'autant qu'il s'agit d'un changement sur le front de détection pour le passage Tx --> Rx

Merci encore pour tous vos travaux à destination de la communauté.

73 - Jean-Roger

armel, 2024-04-25:
Bonjour Jean-Roger,

Quel plaisir d'échanger avec toi. Tu es toujours très clair et méticuleux dans tes messages. En l'occurence, bravo pour le chronogramme. C'est tout à fait explicite. 
Bon, j'avais 1 heure de libre en cette fin de nuit, donc je me suis penché sur la question. Ca n'a pas été complètement trivial à implémenter. Mais je pense (j'espère) être arrivé à un résultat correct.

Tu trouveras en PJ un firmware pour tester. Attention, il est au format zip, Github n'autorisant pas d'ajouter un binaire. Donc, tu devras d'abord le déziper, puis flasher comme à l'habitude. 

Ce que je souhaiterais, si tu en as le temps, c'est que tu testes le plus de cas de figures possibles. Voici quelques scénarios types :

- Appui sur le PTT (le temps que tu veux) puis relâchement, puis appui à nouveau (le temps que tu veux), puis nouvrau relâchement. Ce cas d'usage sera le plus courant. 
- Appui sur le PTT (le temps que tu veux) puis relâchement, jusqu'au TOT,
- Appui sur le PTT (sans relâcher cette fois-ci), jusqu'au TOT.

Je te suggère de régler le TOT à 30 secondes, pour éviter la crampe :)

Tiens moi au courant,

73'

Armel F4HWN.
[f4hwn.packed.bin.zip](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/files/15103710/f4hwn.packed.bin.zip)

F6EGK, 2024-04-25:
Aussitôt dit, aussitôt fait.

Tes trois scénarios procurent bien les résultats escomptés. Ils sont du reste en adéquation avec le comportement relevé dans OpenGD77. Au passage, cet environnement est également doté d'un "décompteur" de temps à l'émission associé au TOT, comme tu l'as implémenté à partir de la version 2.6.

Au trois cas de figure ci-dessus, j'y ai ajouté le quatrième ci-après : 

Appui sur le PTT (temps aléatoire < TOT) puis relâchement, puis appui permanent sur le PTT jusqu'au TOT. C'est également OK.

Bravo Armel !

73 - Jean-Roger

armel, 2024-04-25:
Bonjour Jean-Roger,

Merci pour les tests et content que cela soit conforme au comportement relevé dans OpenGD77.
Et merci pour l'info concernant le décompte. Je l'ignorais. Je possède bien un Radioditty GD77, mais il traine au fond d'un tiroir et je n'ai jamais installé OpenGD77 dessus :) Je ne suis pas très fan des modes numériques. 

Je cloture cette issue.

73'

Armel F4HWN.

closed 2024-04-25 by armel