armel/uv-k5-firmware-custom-feat-F4HWN #60: Bug en Main TX/Dual RX

F6GDN, 2024-04-24:
Bonjour Armel,

Merci énormément pour l'immense travail sur les firmwares UV-K5.
Voici la configuration qui déclenche le bug sur la version 2.6 :
- Paramètre 18 : Slist = All
- Paramètre 21 : ScnRev = Carrier
- Paramètre 24 : F2Shrt = Switch VFO
- Paramètre 26 : M Long = Scan
- Paramètre 53 : Main TX / Dual RX
- Paramètre 62 : SetMet = Tiny
- Paramètre 63 : SetGui = Tiny

J'avais l'habitude d'utiliser mon poste de la façon suivante : 
- Ligne du haut sur l'écran = QRG du relais local (en canal mémoire)
- Ligne du bas sur l'écran = Scan des canaux mémoires des listes I et II
- Je lance le scan de la ligne du bas avec un appui long sur la touche M
- Lorsqu'un appel d'un OM arrivait sur le relais local, un appui court sur F2 me passait de la ligne du bas à la ligne du haut, et je pouvais répondre à l'OM.

Maintenant : 
- Un appui court sur F2 déclenche le scan sur la ligne du haut, m'empêchant de répondre à l'appel.
- Je stoppe le scan sur la ligne du haut pour rechercher le canal du relais, et répondre à l'appel.
- Puis, je ne peux plus relancer le scan, ni en haut, ni en bas, ou, au mieux, le scan se déroule sur la liste I uniquement.
- Je suis obligé d'éteindre le poste et le rallumer.

J'espère que tu pourras reproduire ce bug pour l'étudier ;-)

Merci encore pour ton énorme et excellent travail qui profite à beaucoup de personnes, j'en suis certain.
Tous mes encouragements.
73 QRO !
F6GDN
Eddie

armel, 2024-04-24:
Bonsoir Eddie,

Ca me semble curieux. 
Pour moi, 

- si tu es en mode double VFO (Main TX / Dual RX, ou Dual RX Respond),
- que tu lances un scan sur la ligne du bas,
- que tu fais un appui sur un raccourci (F2Shrt pour toi) pour remonter sur la ligne du haut, 

... alors, il est normal de voir le scan se poursuivre sur la ligne du haut. Ca me semble être le comportement nominal. Et je l'ai toujours vu ainsi. Je me suis même amusé à tester avec une version plus ancienne (en l'occurence, la version 2.0) et c'était déjà comme ca. Je pourrais remonter encore plus loin, mais ca me semble normal. 

En l'état, je reste un peu dans l'expectative ;-)

73' QRO,

Armel F4HWN.

F6GDN, 2024-04-25:
Bonjour Armel !

Merci beaucoup d'avoir répondu si rapidement à mon interrogation.

Je vais examiner mes manipulations, et voir pourquoi j'arrivais au
fonctionnement décrit.

Je ne manquerai pas de te tenir au courant.

Bon courage et au plaisir !
73 QRO !
Eddie
F6GDN

armel, 2024-04-25:
Bonjour Eddie,

Pas de souci. Oui, tiens moi au courant et je regarderais.

73 QRO,

Armel F4HWN.

armel, 2024-04-30:
Clôturé, en attente d'informations complémentaires.

closed 2024-04-30 by armel