armel/uv-k5-firmware-custom-feat-F4HWN #61: BUG d'affichage : démarrage avec port USB de charge

fifisoft57, 2024-04-24:
bonjour

j'ai remarque se soir, quand on démarre le K5 avec la prise  USB branche pour le charge, il y a le  même problème qu'avant avec des pixels a l'écran, je suis en version 2.5

avec l'option affichage complet et BIP au démarrage

cdt
Philippe

armel, 2024-04-24:
Bonsoir Philippe,

Merci.
Oui, c'est exact et je l'avais déjà remarqué. Mais cette fois, je pense que je n'y peux rien :)
C'est plus un problème hardware.
Mais je jetterais un oeil à l'occasion. 

Armel F4HWN.

fifisoft57, 2024-04-25:
bonjour

juste pour info, j'ai regarde avec un K5  sur un firmware d'orogène, cela ne le fait pas

sur le firmware alternatif , cela se fait pendant le bip de démarrage, peut être une piste 

cdt
Philippe

armel, 2024-04-25:
Bonjour,

Oui ok, alors attention parce qu'il y a une subtilité :) 
Ca va aussi dépendre du niveau de ta batterie, je pense.
Je viens de tester à nouveau avec 2 K5 (8) avec un niveau de batterie supérieur à 80% et... rien :) Pas de pixel à l'écran :)

Armel F4HWN.