armel/uv-k5-firmware-custom-feat-F4HWN #63: Offset frequency

p51mustang666, 2024-04-25:
After playing in VFO, i found a trouble when using offset between tx and rx. In uhf offset setting work fine but in vhf qrg, it become crazy. New rx freq  inserted  is reduce of offset amount, and tx is increased of offset amount.
I done same test in uhf , no problem.

Still a pleasure to discover your new versions, very good job Armel !

armel, 2024-04-25:
Hello,

Thank you.
Hmmm, I can't reproduce the problem. Everything works normally on my side. Do you have a specific example to give me?

- VHF frequency used
- Step value (menu 01/63)
- TxODir value (menu 07/63)
- TxOffs value (menu 08/63)

Thank's again :)

Armel F4HWN

armel, 2024-04-25:
Ok, sorry, I can't reproduce this problem.
Take a look at this short video.

73'

Armel F4HWN

https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/127889/1b597967-fcf6-44f5-aec6-6c15f34f6960

armel, 2024-04-25:
Indeed, Reverse mode was actived.

closed 2024-04-25 by armel