armel/uv-k5-firmware-custom-feat-F4HWN #64: Idée pour la 2.7

Chatdurail, 2024-04-25:
Salut Armel, Alex 14UVR267 du groupe FB, 

J'espère que tu vas bien, 

J'aimerai savoir si il était possible de rajouter une option au menu 34 PonMSG : L'idée serait d'avoir le Sound de demarrage et les DEUX lignes de textes programmable dans CHIRP

Cela te semble t'il envisageable ? 

Bien à toi 

73'

armel, 2024-04-25:
Bonjour Alex,

Ok, je pourrais regarder à l'occasion. 
Une idée serait peut-être, en mode PonMSG à ALL, d'afficher ceci.

Cas 1 - Si MESSAGE0 ET MESSAGE1 sont vides dans CHIRP

- VOLTAGE 
- BIENVENUE (après tout, ce firmware est francais)

Cas 2 - Si MESSAGE0 est vide dans CHIRP, mais pas MESSAGE1

- VOLTAGE 
- MESSAGE1

Cas 3 - Si MESSAGE1 est vide dans CHIRP, mais pas MESSAGE0

- VOLTAGE 
- MESSAGE0

Cas 4 - Si MESSAGE0 ET MESSAGE1 ne sont pas vides dans CHIRP

- MESSAGE0 
- MESSAGE1

Ca pourrait être une piste. Par contre, j'ignore combien cela va prendre en mémoire...

Armel F4HWN.

Chatdurail, 2024-04-25:
Merci pour ta réponse. 

Je serai très curieux de savoir le nombre d'octets requis mais en effet un BIENVENUE de base ca serait cocorico pour ton firmware 

bonne soirée

armel, 2024-04-26:
Pour information, Alex, ca coûterait autour de 40 octets... Ca fait un peu cher, sachant qu'il n'en restait plus que 151 depuis la version 2.6. Donc j'hésite un peu...

Armel F4HWN.

Chatdurail, 2024-04-26:
En effet ça fait un peu moins de 30% de ce qui reste... c'est juste une idée :-) au pire garde la dans ton chapeau quand tu fera sauté le broadcast fm...

merci de ton investissement sur ce beau projet