armel/uv-k5-firmware-custom-feat-F4HWN #65: PTT-ID Doesn't survive Power Down

Linden1, 2024-04-25:
Tested on 1*UV-K5 and 2*UK-V5(8) and at many frequecies, FM wide or narrow.
Set PTT-ID to any preset and it works just fine. Turn the radio off then on and PTT-ID is disabled again.
Is this intentional?

Linden1, 2024-04-25:
OK, it's intentional.
I try compiling with ENABLE_DTMF_CALLING=1 and the build fails:
`CMSIS_5/Device/ARM/ARMCM0/Include -c app/app.c -o app/app.o
app/app.c: In function 'APP_TimeSlice500ms':
app/app.c:1622:40: error: 'BEEP_880HZ_200MS' undeclared (first use in this function); did you mean 'BEEP_440HZ_500MS'?
 1622 |                         AUDIO_PlayBeep(BEEP_880HZ_200MS);
      |                                        ^~~~~~~~~~~~~~~~
      |                                        BEEP_440HZ_500MS
app/app.c:1622:40: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:476: app/app.o] Error 1`

armel, 2024-04-25:
PTT ID is per channel setting, you are trying to set it in a channel mode without saving channel settings.
Take a look at ChSave menu to commit your change.

Armel F4HWN.

Linden1, 2024-04-25:
I never saved it, you were correct. Working fine now.

Just the issue of compiling with DTMF Calling?

armel, 2024-04-25:
DTMF calling is no longer supported by this firmware. This is not allowed by amateur radio regulations in many countries. And it took many memory space. There's a lot of #ifdef / #endif conditional code, but I'm not surprised it won't compile.

Armel F4HWN.

closed 2024-04-25 by Linden1