armel/uv-k5-firmware-custom-feat-F4HWN #66: Veille de écran Analyseur de spectre

f4axk, 2024-04-26:
Salut Armel

Ce serait bien d'avoir aussi la possibilité de la mise en veille de l'écran
en mode Analyseur de spectre.

73s laurent

armel, 2024-04-26:
Bonsoir Laurent,

L'analyseur de spectre est un peu une _application dans l'application_. C'est le fruit de l'excellent travail de FAGCI. Et rien n'a été prévu pour mettre en veille l'écran, sauf par une action manuelle. En l'occurence, un appui court sur le bouton F2 te permet de switcher le rétro-éclairage de l'écran, quand tu es en mode analyseur de spectre.

Je pourrais regarder, à l'occasion, s'il pourrait être simple de répondre à ta demande. Mais j'en doute un peu.

73'

Armel F4HWN.

f4axk, 2024-04-27:
Bonjor Armel

Merci pour la réponse ,  j'avais complètement oublié la fonction du bouton F2 qui est très bien expliqué sur
le wiki egzumer donc je considère ma demande nul. 
Le sujet peut être fermé

73'
 laurent F4AXK

armel, 2024-04-27:
Bonjour Laurent,

Et oui, l'analyseur de spectre dispose de ses propres raccourcis. Ca ne m'avait pas sauté aux yeux les premières fois. Ce poste fait tellement de choses, difficile de tout maitriser :) Je m'y perds parfois également.
Je cloture.

Excellent week-end à toi et 73'

Armel F4HWN.

closed 2024-04-27 by armel