armel/uv-k5-firmware-custom-feat-F4HWN #67: Redefine "F" bands range

truefriend-cz, 2024-04-28:
Hi. How i can change "F" band ranger in VFO mode?

armel, 2024-04-28:
Hi. Use case ?

truefriend-cz, 2024-04-28:
In our state, we have different service frequency ranges than the ones that are set as defaults in the firmware. Or I would like to define my own ranges... for example F1 = Trucks (26.8KHz - 28KHz), etc...

armel, 2024-04-28:
All right. Well, this firmware doesn't offer anything to modify them, I'm sorry. But you can do a Scan Range using F+5 in VFO mode. Take a look at this [video](https://www.youtube.com/watch?v=NYsQ3UtbPKs&ab_channel=FredintheShed) if need.

Armel F4HWN.

truefriend-cz, 2024-04-28:
> All right. Well, this firmware doesn't offer anything to modify them, I'm sorry. But you can do a Scan Range using F+5 in VFO mode. Take a look at this [video](https://www.youtube.com/watch?v=NYsQ3UtbPKs&ab_channel=FredintheShed) if need.
> 
> Armel F4HWN.

Yes. I know this function. It's just that adding it under these "presets" using "F1, F2, ..., F7" would be more appropriate. Maybe sometime in the future.

truefriend-cz, 2024-04-28:
CHIRP have problem with your firmware.... 
[image]

truefriend-cz, 2024-04-28:
Do you can add this parameter for modification to CHIRP?

armel, 2024-04-28:
No, as I said, I don't know if it's possible. But if it's possible, this is a job for Chirp. I'll take a look if I have time.  
But as I said too, it's really a specific need. Even if I understand your need, My firmware tries to take into account very generic requests that are useful to everyone.  I can't say yes to everything and everyone :)

f1ixx, 2024-04-28:
Hello
if i understand, you must change table : freq_band_table_t frequencyBandTable[]
This table is in <code>frequencie.cpp</code>
Make modification and compile your new firmware.
To have a personal table via "menu" are to big (i'm think)
Or big job by removing the FM ;-)

73's

Boris