armel/uv-k5-firmware-custom-feat-F4HWN #68: Inconsistent upper band limit while navigating upward/downward

lapinos03, 2024-04-28:
Hi there!

Should the upper band limit be part of the band or not ?
1. Go to VFO mode
2. Enter lower band limit : let's say 137.000 for band F3
3. Let's have step 25KHz (or whatever step)
4. Press DOWN key : 137.000 ->  173.975  (174.000 is skipped, which is normal to me)
5. Press UP key : 173.975 -> 174.000 -> 137.000  (174.000 should not be reached - freq of the next band)

Happens for all bands. Version : 2.6, 2.5, 2.3.

armel, 2024-04-28:
Hi,

According to me, yes, the upper band limit is part of the band. 

Armel F4HWN.

lapinos03, 2024-04-29:
So, here is the culprit : 
https://github.com/armel/uv-k5-firmware-custom/commit/2defc1289d68825e495c45ddcf186fccd14af45a

You included the upper limit as an allowed value when scanning up, before returning to the lower value. But then, for symmetry, when you go below lower value, you should go to the upper value.

To be clear, either it is : 
```	
	if (Frequency >= upper)
		Frequency =  lower;
	else if (Frequency < lower)
		Frequency = FREQUENCY_RoundToStep(upper - pInfo->StepFrequency, pInfo->StepFrequency);
```
or : 
```
	if (Frequency > upper)
		Frequency =  lower;
	else if (Frequency < lower)
		Frequency = FREQUENCY_RoundToStep(upper, pInfo->StepFrequency);
```
To me, a frequency belongs only to one band. Upper limit belongs to the next band. If you input 174000, which band are you on? F4. So you should not reach 174000 when you are on band F3.

armel, 2024-04-29:
Hi,

Ok, this commit fix a bug explain [here](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/issues/11). Of course, I'll not revert it, because it was a real critical bug. As for bands limits, these are closed intervals, always: [50,000, 52,000], [144,000, 146,000], [430,000, 440,000], ... or for PMR [446.00625, 446.19375] (with a step of 6.25), etc. 

Now, about frequency bands, take a look in any transceiver documentation, this is the same thing. For example, here is an abstract of the Yeasy FT-70 bands on RX :

108 - 137 MHz
137 - 174 MHz
174 - 222 MHz
222 - 420 MHz
420 - 470 MHz
470 - 580 MHz

In short, it doesn't really matter. In any case, 2 frequency bands in RX AND TX are never contiguous :) So there's no risk of transmiting on one AND the other at the same time. 

Armel F4HWN.

[image]