armel/uv-k5-firmware-custom-feat-F4HWN #69: RX label on aux channel moves to main channel when PTT on TX-DISABLED frequency

lapinos03, 2024-04-28:
Hi there!

1. Let's have, on main channel, a tx-disable frequency (AM freq for example) (VFO or memory)
2. Let's listen, on auxiliary channel, to a continuous transmission (VFO or memory) : RX blinks on this channel
3. During reception, press PTT: RX moves to main channel, still. At the end of transmission, RX is replaced by >> on main channel.

armel, 2024-04-28:
Hi,

Could you make a short video ?

Armel F4HWN.

lapinos03, 2024-04-29:
https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/assets/891923/97323ed1-2a7b-402c-b0b3-a09bff0995c3

armel, 2024-04-29:
Ok, thank you. It's certainly because VFO A is on TX DISABLE. 
I'm not sure I can simply fix it. But I'll take a look when I have time. 

Armel F4HWN.

armel, 2024-04-30:
Ok. Could you try this firmware and test it ?
I'm not quite sure if my solution is the right one to fix this issue... So test **very carefully**.
You need to unzip firmware before flash it...

Armel F4HWN.
[f4hwn.packed.bin.zip](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/files/15157789/f4hwn.packed.bin.zip)

lapinos03, 2024-05-02:
I will test it as soon as I can (probably during next WE).

lapinos03, 2024-05-04:
So I tested it. It looks to work fine. 👍 
Tested all combinations of (A/B channels) (TD = main TX-DISABLE channel) : 
(MR-TD/MR) (MR-TD/VFO) (VFO-TD/MR) (VFO-TD/VFO) 
(MR/MR-TD) (MR/VFO-TD) (VFO/MR-TD) (VFO/VFO-TD) 

One point of discussion : should it interrupt reception on aux channel (for 4 sec) as the main channel cannot be conversational due to tx-disabling ? Is there a use case where the radio should stay on tx-disable channel for 4 sec ?

Another point I have noticed : 
1. press PTT : TX-DISABLE is displayed immediately BUT RX continues on aux channel
2. release PTT : RX is stopped, TX-DISABLE remains for 2 sec, then another 2s of silence before returning to RX channel.

armel, 2024-05-05:
Fixed in v2.7

closed 2024-05-05 by armel