armel/uv-k5-firmware-custom-feat-F4HWN #71: Cant leave Key Lock when in Hidden Menue

BastelPichi, 2024-05-01:
Thanks for the great work you're doing! I've found a small bug:
When you turn off the radio with KeyLock on, and go into the hidden menue, there is no way to leave the KeyLck. (Long Pressing F doesnt work)

After turning on the radio in normal mode and turning off keylock, everything works fine again.

Suggestion: Either make the F button work if you go into the hidden menu, or just disable keylock when hidden menu gets activated

Or is this a wontfix for memory reasons?

armel, 2024-05-02:
Hi,

Thank you for this issue. It's interesting.

The KEYLOCK feature is specifically designed to prevent changes. In my opinion, it's normal for the KEYLOCK feature to persist after a restart, even if you restart your pocket to access the hidden menus. 

However, you're right: in hidden menu mode, there's no way to UNLOCK the pocket, as you go straight to the menu. This could be a (_small_) problem... 

So I'll try to improve that, if I have enough memory. 

Regards,

Armel F4HWN.

BastelPichi, 2024-05-02:
> Hi,
>
> Thank you for this issue. It's interesting.
>
> The KEYLOCK feature is specifically designed to prevent changes. In my opinion, it's normal for the KEYLOCK feature to persist after a restart, even [...]

Oh and one small thing... It took me a moment to understand what you mean by "pocket". Perhaps "radio" would be a better term?

Dont take this the wrong way, most of us arent native english speakers (me included), and Im trying to say this in the nicest way possible... :upside_down_face: 

And thanks again for the great work youre doing!

armel, 2024-05-02:
Ah ah ok 😂 🤭 Yes, sorry, "pocket" means "pocket radio" / "radio" / "transceiver" 😌

armel, 2024-05-05:
Fixed in v2.7

closed 2024-05-05 by armel