armel/uv-k5-firmware-custom-feat-F4HWN #74: Choix de capacité batterie menu BatType

LiNUT3k, 2024-05-02:
Bonjour,

Dans BatType je n'ai le choix que de 1600 ou 25600 mah, or, j'ai acheté une batterie 3500 mah. Serait il possible d'ajouter cette option ? Merci

Midnight-Cowboy, 2024-05-02:
good idea for diy assembly on 18650 batteries

armel, 2024-05-02:
Bonjour,

Il ne suffit pas d'ajouter un nouveau choix au menu BatTyp. C'est un peu plus compliqué 🤭 Il faut également disposer de données de calibration. Par exemple pour les batteries 1600 et 2200 mAh, voici ces données. Elles permettent de faire une correspondance entre tension et poucentage :

```
const uint16_t Voltage2PercentageTable[][7][2] = {
	[BATTERY_TYPE_1600_MAH] = {
		{828, 100},
		{814, 97 },
		{760, 25 },
		{729, 6  },
		{630, 0  },
		{0,   0  },
		{0,   0  },
	},

	[BATTERY_TYPE_2200_MAH] = {
		{832, 100},
		{813, 95 },
		{740, 60 },
		{707, 21 },
		{682, 5  },
		{630, 0  },
		{0,   0  },
	},
};
```

Sans les données propres à la batterie 3500 mAh, je ne peux rien faire. Et je ne peux pas les inventer 🤗

73'

Armel F4HWN.