armel/uv-k5-firmware-custom-feat-F4HWN #76: BLTime + Tot + F8(BLMax) bug

VoilierRomarin, 2024-05-04:
Bonjour Armel,

1. SetTot [ALL] + BLTime [ON], pas besoin de faire [F][8] car le backlight est allumé en permanence et quand je passe en émission jusqu'à la fin du Tot, le BL se rallume au passage en RX (pas de problème).
2. SetTot [ALL] + BLTime [OOm:05s](ou un autre temps d'extinction) + [F][8] pour allumer le BL en permanence, à la fin du Tot quand le TX se coupe, le BL ne se rallume pas malgré que l'ampoule soit afficher sur l'écran. Il faut appuyer sur une touche pour que le BL se rallume, le même problème se manifeste dans ce cas avec le BLTime [ON].
Je pense que c'est la sortie du Tot avec [F][8] qui pose se dysfonctionnement.

Merci pour ton travail.
Romain

VoilierRomarin, 2024-05-04:
J'ai oublié de préciser le SetPtt[ONEPUSH]

armel, 2024-05-05:
Bonsoir Romain,

Ok bien vu, j'arrive à reproduire le problème. C'est bien un bug.
Je regarde quand j'en ai le temps.
Merci à toi,

Armel F4HWN.

armel, 2024-05-05:
Fixed in v2.7

closed 2024-05-05 by armel